OrcaSlicer/OrcaSlicer #1515: Cant set Top and Bottom Skin direction

Ltek, 2023-07-12:
**Describe the bug**
"Infill Direction" is not setting same direction (as the docs say it will) for top and bottom shells
Either the setting is  broken or that is not what that setting does. 

looking for the Orca equal to Cura (separate settings for Top and Bottom):
"Surface skin pattern and line directions"  *
"The pattern can be set separately to get a different visual effect. If set to lines, the angle of the lines can also be set for more precise control."

Rotating the part or changing the value in setting "Infill direction" doesn't fix - it just changes which side is one way or the other, they are never both the same.

**3mf File for This Bug**
attached in ZIP

**To Reproduce**
use the Infill Direction setting per the screenshot

**Expected behavior**
The "Infill Direction" would work properly per documentation... top and bottom shell directions would be the same per the setting

**Screenshots**
attached

**Printer model**
Sidewinder X2

**Desktop (please complete the following information):**
Win10

[image]
[ALP_CoinDoor_Flap_Final_PLA_MatteBlack.zip](https://github.com/SoftFever/OrcaSlicer/files/12022920/ALP_CoinDoor_Flap_Final_PLA_MatteBlack.zip)

Ltek, 2023-11-06:
Why can't we set the direction?

Ltek, 2023-12-11:
Bueler?

Noisyfox, 2023-12-11:
Use aligned rectilinear.

Ltek, 2023-12-12:
> Use aligned rectilinear. Otherwise the pattern will rotate 90 degrees on each layer.

I do, that is not the fix... I'm putting in a feature request to have a setting Cura has for a LONG time... Top Layer Line Directions

closed 2024-03-19 by github-actions[bot]

Ltek, 2024-03-23:
why not fix this?