OrcaSlicer/OrcaSlicer #1520: [PA Test] - Lines not in center of bed

foreachthing, 2023-07-12:
**Describe the bug**
Test created, does not fit on plate/platter/bed.

**To Reproduce**
Calibration -> PA-Clibration
[image]

**Expected behavior**
Test lines on center of bed so it can be printed.

**Screenshots**
[image]

**Printer model**
Ultimaker 2

**Desktop (please complete the following information):**
 - OS: Windows 10
 - OrcaSlicer 1.6.3

SoftFever, 2023-07-16:
Can you show me how do you define the bed shape?

foreachthing, 2023-07-18:
Please find attached a project file, bed shape (stl) and bed texture (svg).
[OrcaSlicer_PA-Test.zip](https://github.com/SoftFever/OrcaSlicer/files/12079876/OrcaSlicer_PA-Test.zip)

[image]

If you need more information, please let me know.
Thanks!

foreachthing, 2023-11-17:
This is still an issue in 1.8.

closed 2024-03-01 by github-actions[bot]