OrcaSlicer/OrcaSlicer #1524: Avoid crossing wall option ignored intermittently

silentone, 2023-07-12:
Travel moves while printing infill are randomly crossing outer walls when Avoid crossing wall option (with no max detour) is enabled. I tried tweaking many other options and couldn't identify any related setting. Here are screenshots on two layers that are similar. First, it travels around the walls on both ends, then on another layer, it travels across the wall on one end.

[image]

[image]

[wall crossing test 3MF.zip](https://github.com/SoftFever/OrcaSlicer/files/12034105/wall.crossing.test.3MF.zip)

stopanko, 2023-10-08:
Confirmed! This is happening for me only for 0.5 + nozzle size in a settings.   
Have tried Bambu Too. It's doing some circular motions outside with the same setting too.

@SoftFever Could you confirm is this is a Bambu bug or Orca.

[image]
Will add model and Sliced codes in a file.
[travels_crossing_outer_walls.zip](https://github.com/SoftFever/OrcaSlicer/files/12841540/travels_crossing_outer_walls.zip)

closed 2024-01-15 by github-actions[bot]

BlueByLiquid, 2024-07-23:
I have had this same problem