OrcaSlicer/OrcaSlicer #1527: Unable to 'Send' job to printer SD card [The printer is required to be in the same LAN]

escfoe2, 2023-07-13:
**Describe the bug**
Unable to 'Send' job to printer SD card.
Dialog says: The printer is required to be in the same LAN as Orca Slicer

**3mf File for This Bug**
Any

**To Reproduce**
Slice a model, change 'Print Plate' dropdown to 'Send' and try to send to your device.

**Expected behavior**
It should send the file to the printer and the printer will then save it to its SD card.
Works as expected in Bambu Studio.

**Screenshots**
[image]

[image]

[image]

[image]

**Printer model**
Bambu X1 Carbon

**Desktop (please complete the following information):**
 - OS: Windows
 - Version 11

kylek29, 2023-07-20:
Out of curiosity, are you on the same network as the printer, and it is still showing this error message?

I tried it myself with BambuStudio and get the same error message in both when not on the same network. I was hoping they adjusted the ability to Send (but not print) over the cloud as the work-around of doing a "print" and then cancelling that print is annoying.

lbibass, 2023-08-27:
I am having the same issue. I am on an enterprise network.

jamesbrindle, 2023-09-16:
I have the same issue. version 1.70... I've made sure the firewall isn't blocking anything, and the official Bambu Studio works fine (recent upgrade).

escfoe2, 2023-09-16:
Alright ladies and gents, I think I know the problem.  The application needs re-connected to your printer.  The whole step through thing where you enter in the access code that's one the printer, yeah that.  It randomly popped up one day in my slicer and when I put in the details, it worked like a charm.  So the issue is that the access codes are out of sync but Orcaslicer doesn't know that or doesn't think to tell you that.

jamesbrindle, 2023-09-17:
Is this is LAN Only mode? How would you get the prompt up to enter the access code?

escfoe2, 2023-09-17:
> Is this is LAN Only mode? How would you get the prompt up to enter the access code?

No.  It's not LAN only mode.
And I wish I knew.

MxEdwrd, 2023-11-30:
Hi folks. I had this same issue after just getting my new P1S and wanting to use Orca Slicer. After searching for solutions on the problem I came across this thread. Using the below process, I was able to get the "send" option to work within Orca Slicer. I am on the latest build (1.8.0).

[image]

First, go to the "Device" pane and locate the machine name. Click on it and you should see a drop down. Click on the box with arrow pointing to the right to log out of the printer. You will then get a pop-up asking if you would like to log out, click confirm.

[image]

[image]

Once logged out, you should save whatever you were doing within the slicer and fully close the software (quit or command + Q on MacOS). Then, reopen the software. Once its opened, ensure you are logged in with your account. You can check this by going to the home page and verifying if there is a "Log out" button below your name and profile picture.

[image]

Go to the device pane once again and it should say that there are no printers hooked up. Click on where it says "No Printer" and it should bring up a window with "My Device" and "Other Device" listed. If you are correctly logged in, you should see your printer's original name in the other device section. 

[image]

[image]

Click on your printer's name and it will ask if you would like to log in to the printer with the current account. You will have to accept the terms and conditions as well as the privacy policy. Once that is accepted, you can click "Confirm" and it will log you back into the machine. Once successfully logged in, a window will pop up saying "Log In Successful."

[image]

[image]

Now, load up a project or insert a model, slice it, and then attempt to use the "send" option under the print drop down. The "Send to MicroSD Card" window should now appear and the send button should no longer be greyed out telling you that it is unable to send. Click on the send button and the file will then be sent to the printer's MicroSD card. 

[image]

As I stated before, this process worked for me but your mileage may vary. Thanks and good luck printing!

escfoe2, 2023-12-01:
I actually just had this problem in BambuStudio as well. I think the driving factor for me is my computer goes through a networking switch before getting back to the router, whereas my printers connect directly to the wifi.  I followed MxEdwrd's steps, and then also turned off my firewall (not recommended), and it worked.

closed 2024-03-09 by github-actions[bot]