OrcaSlicer/OrcaSlicer #1528: Add in Vertical Shrinkage in Filament Settings for printers where mm/step are non-accessible (ex. Bambu Lab printers)

BikeGuy1992, 2023-07-13:
So far after running through the calibration tests for Orca Slicer I was getting good results printing with PyroFlex TPU. However there is one glaring issue that I came across. When doing the Orca Tolerance Test, and you actually measure the part when it is done everything on it (in my case) is coming out 0.3mm small. 6mm is more like 5.7mm. the instructions for the tolerance test do not mention anything on how to account for small or big parts or what settings to change when being printed off, period. Printed off a battery box for my FPV quadcopter and it's super difficult to get the battery out because of it. The box fits up to the bottom plate no problem since it has tabs that mate up with the frame, but getting the battery in and out is almost next to impossible. 

What would normally happen with any other printer to correct for parts out of spec is you would go in and change your mm/steps in the machine to help get the part in the ballpark, then use the slicer to account for the last little bit of compensation. However, when poking around in the Bambu Lab X1C printer there is no option to change mm/step. So I went searching in Orca Slicer 1.6.3 to see if there was anything and 2/3 of my problem got fixed while the other 1/3 didn't. There is a 'shrinkage' setting in the filament settings page that can be adjusted to help account for parts coming off the printer that are either too small or too big. But, this only works for X and Y, not for Z. 

Based on the shrinkage offset, I'm proposing for a feature request that also includes a vertical shrinkage in the Z axis as an option in the filament settings menu. That way you can adjust it per filament being used and it can be saved. It would work the same way as the current shrinkage setting but in the Z axis only (ex. part measures 94mm tall when it is supposed to be 100mm, you input 94% to account for it and the part comes out being 100mm tall). 

Sure, there is the scale in the build plate menu, but honestly this is a horrible way to account for small or big parts in any of the three axis. You can't save that per filament type, it resets every time you start with a new build plate and I really do not want to have to memorize or write down what scale I need to put in each time. The other way would be change X-Y compensation but this only enlarges or shrinks the part down in all axis, and this is only save for print settings, not based on filament type. 
[image]
[image]
[image]
[image]
[image]
[image]
Alternatively the other way would be change your CAD drawing to account for this. But honestly this is even far worse of a solution than changing the scale in the build plate area because you should be able to o draw your parts up in CAD, send it straight to the printer, and get dimensionally accurate parts every time. Sure, everything does not come out EXACTLY perfect but if it's with 0.01mm or less I'm fine with that. Bit when parts are coming out 0.2mm or greater that is not acceptable. And having no way to control in any axis (either by changing mm/step or within the slicer or only being able to change two out of three axis) that's not okay. 

So yeah, this would be a nice and useful feature to have to account for the lack of not being able to change mm/step in the Bambu Lab printers. Cura Slicer can account for this with horizontal expansion and horizontal hole expansion already, and Marlin based printers you can go in and adjust the mm/step no problem. Pretty sure Prusa printers you can do the same as well.

closed 2023-11-22 by github-actions[bot]

eloo, 2023-12-31:
can we reopen this?

i'm also facing a bit of z shrinkage using ABS+ so an option to compensate zshrink for specific filaments would be great

clc3123, 2024-04-30:
I always scale up z dimension a little bit less than x & y dimension.
So it would be nice to have a z axis shrinkage setting in filament tab.

lxlolxl, 2024-05-04:
I had the same problem when printing ABS, which was one of my problems when I moved from CURA to Orca, and I think that the addition of Z-shrink compensation to the line is a necessary option.