OrcaSlicer/OrcaSlicer #1534: Make Overhang Printable Feature [$300 Bounty]

odie5533, 2023-07-14:
Cura has feature called Make Overhang Printable in their CuraEngine. Cura includes two adjustment settings: Maximum Model Angle (conical_overhang_angle) and Maximum Overhang Hole Area (conical_overhang_hole_size).

Cura ConicalOverhang algorithm: https://github.com/Ultimaker/CuraEngine/blob/main/src/ConicalOverhang.cpp

The setting must be used judiciously, but it saves having to use supports or modify the model. It has a wide range of uses from organics like statues, functional prints, hard shapes, and soft shapes. As long as the changes are subtle enough or in unimportant places, it can be a godsend since it avoids supports, saving print time and cleanup.

Knob: https://www.thingiverse.com/thing:665410/files
Yoda: https://www.thingiverse.com/thing:276994/files

odie5533, 2023-07-17:
@Noisyfox This is incredible!!! I can't believe it's working! You are amazing!!!!!!!

I've added settings for the implementation and put up a PR #1556 

Would you be able to add the hole_size stuff to the code? The code seems a little unfinished in its current state so not sure how to clean it up exactly.

Noisyfox, 2023-07-18:
> Would you be able to add the hole_size stuff to the code? The code seems a little unfinished in its current state so not sure how to clean it up exactly.

Yeah I will, and the code needs more testing too because I don't fully understand how libslic3r works either :p

Noisyfox, 2023-07-19:
@odie5533 I'v done implementing the hole size part.

odie5533, 2023-07-19:
> I am also thinking of adding this setting to modifiers, so you can apply this only to part of the model. Though this might be tricky to implement as it will make calculating the overhang support shape for different parts more complicated.

If this would be possible it would be really cool and a feature that even Cura doesn't have! Let me know what you think.

Noisyfox, 2023-07-20:
The main problem is I couldn't figure out how the algorithm work if say the user added multiple modifiers to the same layer with different max overhang angle, since the polygon offset function cannot apply different offset values on different directions.

On the other hand, I think it's totally pointless to allow specifying different max angle value to different part of the model, what matters most is the ability to turn on & off the conical overhang feature on demand, the max angle and hole size configuration can be a shared setting for all places where this feature is enabled, which make the calculation much simpler. However, this makes the configuration UI a bit tricky to design, for example, do we want a global switch to turn this feature on, then an option in the modifier to turn it off? Or do we want the opposite? Or both? And where to put the shared max angle/hole size settings?

Any suggestion will be appreciated.

foreachthing, 2023-07-20:
@Noisyfox 
my 2¢:
* "globally" enable/disable M.O.P.F. and set angle.
  * I'd put it in "others" -> "Special mode".
* have modifiers to be able to _disable_ M.O.P.F.
* nice-to-have: enable/disable it per part

If you enable M.O.P.F., you might want to turn it off somewhere locally. No need to enable it on a "small" section, I think.

Noisyfox, 2023-07-21:
> you might want to turn it off somewhere locally. No need to enable it on a "small" section, I think.

I think you are right, as there should be more overhangs than bridges for models which the M.O.P.F. is useful and it shouldn't be too difficult to apply modifiers to those bridges.

> * nice-to-have: enable/disable it per part

This is already supported btw, you can change this setting per object.

Noisyfox, 2023-07-22:
[image]
[image]

Now you can disable this per modifier.

odie5533, 2023-08-02:
PR #1615 closes this

Bounty paid to @Noisyfox and also made a donation on ko-fi to @SoftFever 

Thank you both so much for working on this feature!

closed 2023-08-02 by odie5533

rjmessing, 2023-11-25:
I've run into a problem with the make overhangs printable.  If I enable it for some parts any internal geometry disappears.  This part of a spool winder I'm making, the overhang is pretty steep and I thought I'd try making them printable.  As you can see the central area for the screw gets removed.

[image]
[image]

triangleman83, 2023-12-03:
I am working on a model that has some challenging overhangs and this feature is improving its printability greatly. I am also coloring a few layers at the top of the model and I found that having this option enabled caused the slicer to create too many toolpaths through the colored portion all the way down through the model. This led to needless color changes throughout the infill which was now colored walls. I found that creating a modifier over the colored area to disable the feature solved the issue. I did not think this necessitated a bug report myself since it could just be two conflicting orders given to the slicer and also since the modifier solves the issue. I can provide more information if needed but wanted to make this comment for posterity in case anyone else has trouble with coloring with this option enabled. If the coloring is also required on the areas being modified, that may not be solvable with a modifier though. Thank you for the great feature!