OrcaSlicer/OrcaSlicer #1538: [Serious bug] Top layer not being treated as top layer.

Rhotix, 2023-07-14:
**Describe the bug**
As shown in the picture only the last layer is shown and treated as top layer. So it will use a different flow for it. From 10 top layers there are 9 treated as solid infill. This happens to many parts. Sometimes there are 5 top layers and 5 solid infill layers and so on. I need different settings for the top layer because otherwise scars occure in my prints. The scars happen because the Slicer interrupts the loop at its end, even with 0% seam gap. Getting a higher flow to counter this, results in rough and ugly top layers if there are too many solid infill layers. Also the print speed is affected by this behaviour, which also causes sometimes big problems on complex parts.

**To Reproduce**
Different behaviour on different parts

**Expected behavior**
If I e.g. set up 10 top layers , the last 10 layers should be treated like that. 

**Screenshots**
[image]
[image]

**Printer model**
X1C

Annuit20, 2023-07-16:
Hey, there is always on one Top surface layer! Other solid layers are bottom or top solid infill layers. Other infill (with space) is sparse infill. So nothing wrong.
If you like you can set settings for top surface and solid infill the same and get what you want.

Dinchy87, 2023-07-18:
I have reported maybe the same bug. and for me it is a bug because Bambu Studio does not behave like this and if I print this way I see holes all around my prints

https://github.com/SoftFever/OrcaSlicer/issues/1560

closed 2023-11-22 by github-actions[bot]