OrcaSlicer/OrcaSlicer #1540: After slicing using filament "1", temperature in graphicical display show 0 instead of temp

34bwood, 2023-07-15:
**Describe the bug**
When using filament 1, the temp shows 0 when I change Line Type to Temperature.  If I make first layer different from other layers, First layer displays 0, other layers shows correct.  The Gcode is created with the correct temp, it just doesn't show in the graphics display correctly.

**3mf File for This Bug**
See attached

**To Reproduce**
Slice a file using Filament 1.  View the temperature in the graphical view.  Filaments in other slots work as expected.

**Expected behavior**
Temperature(s) shown in graphical display should match settings set in filament.

**Screenshots**
See attached

**Printer model**
Voron Trident 300

**Desktop (please complete the following information):**
 - OS: Mac OS 
 - Version 12.6 M1 chip
[Orca_Slicer_Bug.zip](https://github.com/SoftFever/OrcaSlicer/files/12061222/Orca_Slicer_Bug.zip)

BJam, 2023-07-19:
I had the same issue after removing `M109 S[nozzle_temperature_initial_layer]` from my Machine start G-code.  Adding it back made the slicer display the correct first layer temperature.

34bwood, 2023-07-19:
I do have Start layer info, but it's a little different for a voron.  Plus, it works just fine for filament 2, 3, 4, etc.  Just not filament 1.  I can even make filament 1 and 2 the exact same filament and 2 works, 1 doesn't.  Here's my start G-code:

M104 S0 ; Stops OrcaSlicer from sending temp waits separately
M140 S0
print_start EXTRUDER=[first_layer_temperature] BED=[first_layer_bed_temperature] CHAMBER=[chamber_temperature]

closed 2023-11-22 by github-actions[bot]