OrcaSlicer/OrcaSlicer #1545: Weird behaviour of "Detect overhang walls"

zviratko, 2023-07-15:
**Describe the bug**
I don't even know what the setting "Detect overhang walls" in the Quality tab is supposed to do now, as it seems to collide with "Slow down for overhangs" in the Speed tab.
Presets had it enabled and I haven't touched it.

Yesterday I was  printing Hydra AMS mod and noticed that one part wasn't looking quite right when printing (see pictures). It looked weird both sliced and printed, so I investigated and found out that disabling this setting fixes it.

Enabling it seems to speed up printing (or at least claims to) by a bit, both with "Slow down for overhangs" enabled or disabled, which makes little sense, I think.

**To Reproduce**
1) Download https://files.printables.com/media/prints/392134/stls/4144758_009d523e-aa5f-49db-9aa6-ea240e40bbe7/hydra-ams-middle-107.3mf and slice it with "Detect overhang walls" enabled
2) Look at the left part and notice that the overhang (or rather reverse-overhang) part looks weird
3) Disable "Detect overhang walls"
4) Reslice and compare

Playing with other settings makes those weird layers shift a bit or appear in different places.

[image]
[image]

**Expected behavior**
I have no idea what this setting does, should it just be removed?

**Printer model**
X1C

**Desktop (please complete the following information):**
 - OS: macOS
 - OrcaSlicer Version: yesterday's action build or such (1.6.4-dev)

SoftFever, 2023-07-16:
It seems fine on my machine
[image]

Can you share the project with settings too?

zviratko, 2023-07-16:
Sure, here you go 
[hydra-ams-middle-1.07-bug.zip](https://github.com/SoftFever/OrcaSlicer/files/12063572/hydra-ams-middle-1.07-bug.zip)

uberjay, 2023-07-24:
I reproduced this -- the artifact appears to only be present when using the Arachne wall generator _and_ "Detect overhang walls" enabled. Setting the wall generator to Classic, _or_ disabling Detect overhang walls fixes the issue. (It also affect the current -- 1.08 -- AMS Hydra middle piece model.)

FWIW, these models both trigger the non-manifold edge warning. Unsure if it's related or not. (mostly because i have no idea what this warning means!)

sudokrang, 2023-07-31:
I also had this issue with 1.08 on all parts. Reprinted with classic ( Disabled Arachne ) and had no issues.

zviratko, 2023-07-31:
It's not the only model that does it, I just I ignored it in the past. The error seems to be minor so it only shows on big surfaces (although I think I noticed it on some round fidget spinners first). It's possible it's caused by broken models, but... there are many :)

SoftFever, 2023-08-01:
> Sure, here you go [hydra-ams-middle-1.07-bug.zip](https://github.com/SoftFever/OrcaSlicer/files/12063572/hydra-ams-middle-1.07-bug.zip)

Thanks.
I'm able to reproduce it with this project.
It's kinda random. re-slice the model a few times, the result will change each time

discip, 2023-10-04:
@SoftFever
This seems to be solved at least with the [latest build](https://github.com/SoftFever/OrcaSlicer/actions/runs/6395503184) regardless if fixed or unfixed model, both with **`Detect overhang walls`** enabled and disabled.

@zviratko
If this is fixed for you, please don't forget to close. 😊

closed 2023-10-04 by SoftFever