OrcaSlicer/OrcaSlicer #1546: K1 Max profile

Myklt8, 2023-07-15:
**Is your feature request related to a problem? Please describe.**
No

**Describe the solution you'd like**
Add K1 and K1 Max to the Creality section, hopefully with G-codes that will continue using the advanced features of these machines

**Describe alternatives you've considered**
Only Creality Print, but would like to continue using Orca

**Additional context**
My printer is still a couple of weeks out, and I will still be using my Ender 3. just would like to use the same slicer for both
I am just not good at G-code (yet)
Thank you guys for all of your work

SoftFever, 2023-07-16:
Will only consider it once they have complied with the GPL license, which includes admitting that they use Klipper/Moonraker/Fluidd and releasing the source code.

xand3r40r93, 2023-07-25:
@SoftFever great, because that's what they want) so that as many people as possible use only software from Creality

SoftFever, 2023-07-25:
Thanks all.
I'm closing this for now.

FFengIll, 2023-09-10:
I see here is a profile and I used it well.
FYI.

> https://www.printables.com/model/537828-k1k1-max-orcaslicer163-config/files