OrcaSlicer/OrcaSlicer #1552: Bed temperature of first layer carrying on to the rest of the layers

Glenn1963, 2023-07-16:
### Discussed in https://github.com/SoftFever/OrcaSlicer/discussions/1492

<div type='discussions-op-text'>

<sup>Originally posted by **Studioscale** July  8, 2023</sup>
i set the slicer up so that the first layer would be 120 and the rest 110 but its printing the whole thing at 120. The extruder is doinf it correctly, reducing from 250 to 240 for the second layer on, but not the bed temp. Am I doing something wrong here?
[image]
[image]
[3DBenchy_ABS_43m38s.txt](https://github.com/SoftFever/OrcaSlicer/files/11990942/3DBenchy_ABS_43m38s.txt)
</div>

Glenn1963, 2023-07-16:
As I described in #1492 , changing the Layer change machine G-code seems to disable generating the bed temperate change M140 code. Just changing the Other layers temperature will allow the code to be generated normally again.

closed 2023-11-22 by github-actions[bot]

btCRO, 2023-12-15:
And today I found that disabling automatic brim would forgot to set M140 at second layer in 100% of situation I'we tried.
Yesterday I found if you change speed without saving and reloading M140 is missing from secon layer.