OrcaSlicer/OrcaSlicer #1555: Fluidd Camera feed on Device Tab Flickers when not Scrolling and goes Black when Scrolling

ChaosBlades, 2023-07-17:
**Describe the bug**
If you add a printer with Fluidd and it has a camera that camera will flicker. If you scroll the page the camera goes black until you stop scrolling then it goes back to flickering.

**To Reproduce**
Add a printer running Fluidd with a camera configured on the dashboard and open the Device tab in OcraSlicer

**Expected behavior**
The camera to not trigger photosensitive epilepsy.

**Screenshots**
I would need a video to show it but just imagne the video stream flashing to black a couple of times every second.

**Printer model**
Voron 2.4

**Desktop (please complete the following information):**
 - OS: Pop!_OS 22.04 LTS x86_64 
 - Version 1.6.3

ChaosBlades, 2023-07-18:
After some testing this is a Linux specific issue. Not sure what HTML rendering engine the Linux version of OrcaSlicer uses but it clearly has issues.

Dfwsb2, 2023-07-18:
If you are using LED lights in your voron have you tried adjusting your camera settings in linux? It's would be something like this depending on your camera. It has to do with the cam running at 60hz and the freq of the leds. I had the same issue and this fixed it v4l2-ctl --set-ctrl=power_line_frequency=1

ChaosBlades, 2023-07-19:
No change using that command.
Does not matter if I use lights in enclosure, room lights, or an LED flashlight.
No issues on any other browser. Only flickers within OrcaSlicer.

The interface within OrcaSlicer is also very choppy when scrolling and responds slow when interacting with it.

closed 2024-02-20 by github-actions[bot]