OrcaSlicer/OrcaSlicer #1558: Extrusion Multiplier for each Feature Type / Line Type

arshish1612, 2023-07-18:
**Is your feature request related to a problem? Please describe.**
Unable to control flow rates / extrusion multipliers for different feature types. Currently you can only adjust the default multiplier and the top Solid Infill. However there is a need to set the flow rate for different layers. This is available in Cure and is extremely useful in getting prints perfectly dialled in.

**Describe the solution you'd like**
Using Cura as a reference, a section to se the extrusion multiplier for different line features.
[image]

**Describe alternatives you've considered**
In PrusaSlicer you have a substitute GCode which is OK but Cura implements it better

**Additional context**
Add any other context or screenshots about the feature request here.

The760, 2023-07-21:
Would absolutely love to see this.

arshish1612, 2023-07-26:
@lukasmatena Would you be aware if anyone has had a look at this pls.

arshish1612, 2023-07-28:
Bump Bump Bump

arshish1612, 2023-08-29:
Any likelihood of seeing this?

closed 2023-12-06 by github-actions[bot]

lambonick, 2023-12-11:
> Please open this issue. Its absolutely necessary

I 2nd that!

arshish1612, 2023-12-12:
I wonder why this has not even been acknowledged by the devs as yet..

sonmoku, 2024-02-04:
Necessary feature for tweaking lightweight mechanical parts, lowering gas permeability and other cases

Please consider it - Love the Slicer, Thank you!

arshish1612, 2024-02-05:
Why is this Closed? Unless the Status os Opened again no one is going to see our comments to this. Should we create a new Issue?

arshish1612, 2024-02-05:
Ive created a new issue so this gets looked at: 
https://github.com/SoftFever/OrcaSlicer/issues/3977