OrcaSlicer/OrcaSlicer #1560: Second and First plateau Top Layer

Dinchy87, 2023-07-18:
I have a strange if I slice a Stair or anything that has multiple Top Layers on different Heights the second an all above Top Layer I see the inner wall  so it makes not a top layer like it should.

And I cannot see any option to turn this off. In the Bambu Studio slicer it slices the file and each Top layer as it should.

What am I not seeing?!

Both are sliced on a Macbook Pro M2

here is an example of Orcaslicer
[image]

and here an example of Bambu Studio

[image]

Do you see the inner wall rings there? I think they should not be there.

Here is also the 3mf file of this scene.
[StairTest.3mf.zip](https://github.com/SoftFever/OrcaSlicer/files/12086243/StairTest.3mf.zip)

jeremytodd1, 2023-07-19:
Issue is being worked on in issue 582. 

Please close. In the meantime, disabling Arachne will get your single perimeters for top layers again.

Noisyfox, 2023-07-22:
> In the Bambu Studio slicer it slices the file and each Top layer as it should.

If you look more closely you'll notice that when Arachne is selected the "Only one wall on top surfaces" options is disabled in BBS.

closed 2023-11-22 by github-actions[bot]