OrcaSlicer/OrcaSlicer #1565: Outer_Only Brim also encompassing insides

arshish1612, 2023-07-19:
**Describe the bug**
When you select Outer_Only brim, the inner areas also get covered by the Brom 

**3mf File for This Bug**
3mf is attached
[BrimIssue.3mf.zip](https://github.com/SoftFever/OrcaSlicer/files/12096515/BrimIssue.3mf.zip)

**To Reproduce**
Use the attached file and select "Outer_Only: Brim. The inner sections are not excluded form the Brom

**Expected behavior**
Inner sections should not get covered by the Brim

**Screenshots**
Attached:
[image]

**Printer model**
FlSun V400

**Desktop (please complete the following information):**
 -Macbook Air 13 M1 - 13.4.1 (c)

Noisyfox, 2023-07-22:
I think the problem is that there is a very thing opening on the side of the hole which makes it not really an inner area.

arshish1612, 2023-07-24:
But this does not happen with PrusaSlicer... So something is not right

Noisyfox, 2023-07-25:
The key is the value of the "Slice gap closing radius", which is 0.049 in PrusaSlicer by default, which will remove the crack and make that hole an inner area. If you change this settings to 0 (which is BBS/Orca's default value) you will have brim in that area too:
[image]
[image]

On the other hand, if you set that value to 0.049 in BBS/Orca then it will do what you want:
[image]

arshish1612, 2023-07-25:
Awesome. Thanks for the fix.

closed 2023-11-22 by github-actions[bot]