OrcaSlicer/OrcaSlicer #1570: Multiple filament slots does not change the actual filament settings

Naninani, 2023-07-21:
**Describe the bug**
When creating a new profile out of the predefined ones and change some setting is the "Settings Override" section, the new profile settings override the ones in the existing system presets. This happens only if there are multiple Filaments slots added from the "Add one filament" option.

**3mf File for This Bug**
If it is related to slicing, please append the 3mf file. It could be extremely helpful to solve the issue.

**To Reproduce**
1. Backup all existing profiles.
2. Wipe all profiles.
3. Logout if logged in (issues repo both logged in or out)
4. Uninstall Orca.
5. Install again.
6. Select Generic PETG.
7. Edit the Settings Override of Filament settings.
8. Set Z hop type to Slope
9. Set the Wipe Distance to 10mm
10. Save present to new one with name 1.
11. Add Primitive Cube
12. Add additional filament slot.
13. In slot 1 chose filament 1, in slot 2 chose Generic PETG
14. Slice with 1 and enable the Travel and Wipe to see the lines.
15. Set to slot 2 with the Generic PETG and slice again.
16. What happens is that it does not actually switch to slot 2 settings, but choses the settings from slot 1.
17. This happens only if there are more than 2 added filament slots. If you chose Generic PETG from slot 1, the settings are correctly updated.

**Expected behavior**
It should correctly update the settings for the different filament slots.
This is an extremely nasty bug that I managed to reproduces in both Orca and Bambu slicers. It actually ruined tons of prints for me and I spent countless hours figuring out why my prints are so off when using different filament slots.

**Screenshots**
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

**Printer model**

**Desktop (please complete the following information):**
 - OS: [Mac]
 - Version [1.6.3]

psiberfunk, 2023-07-21:
If this replicates in upstream, is it logged there too ?  if so, please link here, or log it there as well ?  It would be easier for @SoftFever if Bambu fixed this and he pulls it in IMO.

Naninani, 2023-07-22:
https://github.com/bambulab/BambuStudio/issues/2098

Naninani, 2023-08-17:
@SoftFever are you able to repo this on your end? This seems to be a rly bad bug.

Naninani, 2023-10-27:
@SoftFever, this issue has been fixed in Bambu Studio: 1.7.6.92. Do you plan to port the latest changes?

SoftFever, 2023-10-28:
> @SoftFever, this issue has been fixed in Bambu Studio: 1.7.6.92. Do you plan to port the latest changes? [image]

Hi 
Can you try the latest V1.8.0 beta release? Let me know if the bug presists.

Naninani, 2023-11-20:
@SoftFever its fixed - I tested the latest 1.8 release. Closing this bug.

closed 2023-11-20 by Naninani