OrcaSlicer/OrcaSlicer #1576: BambuSource.dll crashing Camtasia Snagit 2023

VeryKross, 2023-07-22:
**Describe the bug**
Even without Bambu Studio running, the BambuSource.dll is somehow interacting with Camtasia Snagit 2023 and causing it to crash after Snagit has been running for 5min or so.  I've seen similar posts here regarding BambuSource.dll appearing to affect other applications as well (Microsoft Teams, etc.) but so far, I am only seeing its effect on Snagit.

**To Reproduce**
I have reinstalled both Snagit 2023 and the latest build of Bambu Studio (1.07.00.65). 
1. After starting Snagit, observe that the Snagit "widget" is visible (assuming you have that option turned on)
2. After 5min or so, you'll notice the widget is gone
3. Check the Windows Application log and find an error report similar to:

        Faulting application name: SnagitCapture.exe, version: 23.2.0.30713, time stamp: 0x64ad9c74
        Faulting module name: BambuSource.dll_unloaded, version: 0.0.0.0, time stamp: 0x64a7650e
        Exception code: 0xc0000005
        Fault offset: 0x000000000008c361
        Faulting process id: 0x0xA3BC
        Faulting application start time: 0x0x1D9BCC284C2AC06
        Faulting application path: C:\Program Files\TechSmith\Snagit 2023\SnagitCapture.exe
        Faulting module path: BambuSource.dll
        Report Id: 95869223-736c-49c7-a5c0-d332115af0f3
        Faulting package full name: 
        Faulting package-relative application ID: 

**Expected behavior**
There should be no interactions between Bambu Studio and any other software on the computer, especially when Bambu Studio isn't even running. This did not happen with Bambu Studio prior to the beta.

**Printer model**
P1P

**Desktop (please complete the following information):**
 - OS: Windows 11

chrdavis, 2023-08-13:
According to issue #1748 this is fixed in 1.7.2

JDProfresh, 2023-08-13:
@chrdavis You linked a UKR localization commit merged 3 days ago. How does this address this issue? Also, 1.7.2 isn't a build version in this project - unless I'm just being silly!

chrdavis, 2023-08-14:
Oops.  I was referring to Bambu Studio: https://github.com/bambulab/BambuStudio/issues/1748

It has the same issue and is fixed in Bambu Studio 1.7.2

JDProfresh, 2023-08-16:
I was experiencing this same issue, and so far have experienced no crashes in OrcaSlicer using Beta-3 and 1.7.2 BambuStudio. Seems like it may have been addressed..

closed 2023-11-22 by github-actions[bot]