OrcaSlicer/OrcaSlicer #1577: P1S config support

handmade0octopus, 2023-07-22:
Hi,

I tried several ways, copying BBL folder and other from BambuSlicer but I cant get it to recognize the printer configs.

I also tried opening 3MF from BambuSlicer with preset saved but I am not able to save it as OrcaSlicer just closes during procedure. Although it creates the preset but it doesn't get loaded.

```json
{
    "auxiliary_fan": "1",
    "bbl_calib_mark_logo": "1",
    "bed_custom_model": "",
    "bed_custom_texture": "",
    "bed_exclude_area": [
        "0x0",
        "18x0",
        "18x28",
        "0x28"
    ],
    "before_layer_change_gcode": "",
    "change_filament_gcode": "M620 S[next_extruder]A\nM204 S9000\n{if toolchange_count > 1 && (z_hop_types[current_extruder] == 0 || z_hop_types[current_extruder] == 3)}\nG17\nG2 Z{z_after_toolchange + 0.4} I0.86 J0.86 P1 F10000 ; spiral lift a little from second lift\n{endif}\nG1 Z{max_layer_z + 3.0} F1200\n\nG1 X70 F21000\nG1 Y245\nG1 Y265 F3000\nM400\nM106 P1 S0\nM106 P2 S0\n{if old_filament_temp > 142 && next_extruder < 255}M104 S[old_filament_temp]{endif}\nG1 X90 F3000\nG1 Y255 F4000\nG1 X100 F5000\nG1 X120 F15000\n\nG1 X20 Y50 F21000\nG1 Y-3\n{if toolchange_count == 2}\n; get travel path for change filament\nM620.1 X[travel_point_1_x] Y[travel_point_1_y] F21000 P0\nM620.1 X[travel_point_2_x] Y[travel_point_2_y] F21000 P1\nM620.1 X[travel_point_3_x] Y[travel_point_3_y] F21000 P2\n{endif}\nM620.1 E F[old_filament_e_feedrate] T{nozzle_temperature_range_high[previous_extruder]}\nT[next_extruder]\nM620.1 E F[new_filament_e_feedrate] T{nozzle_temperature_range_high[next_extruder]}\n\n{if next_extruder < 255}\nM400\n\nG92 E0\n{if flush_length_1 > 1}\n; FLUSH_START\n; always use highest temperature to flush\nM400\nM109 S[nozzle_temperature_range_high]\n{if flush_length_1 > 23.7}\nG1 E23.7 F{old_filament_e_feedrate} ; do not need pulsatile flushing for start part\nG1 E{(flush_length_1 - 23.7) * 0.02} F50\nG1 E{(flush_length_1 - 23.7) * 0.23} F{old_filament_e_feedrate}\nG1 E{(flush_length_1 - 23.7) * 0.02} F50\nG1 E{(flush_length_1 - 23.7) * 0.23} F{new_filament_e_feedrate}\nG1 E{(flush_length_1 - 23.7) * 0.02} F50\nG1 E{(flush_length_1 - 23.7) * 0.23} F{new_filament_e_feedrate}\nG1 E{(flush_length_1 - 23.7) * 0.02} F50\nG1 E{(flush_length_1 - 23.7) * 0.23} F{new_filament_e_feedrate}\n{else}\nG1 E{flush_length_1} F{old_filament_e_feedrate}\n{endif}\n; FLUSH_END\nG1 E-[old_retract_length_toolchange] F1800\nG1 E[old_retract_length_toolchange] F300\n{endif}\n\n{if flush_length_2 > 1}\n; FLUSH_START\nG1 E{flush_length_2 * 0.18} F{new_filament_e_feedrate}\nG1 E{flush_length_2 * 0.02} F50\nG1 E{flush_length_2 * 0.18} F{new_filament_e_feedrate}\nG1 E{flush_length_2 * 0.02} F50\nG1 E{flush_length_2 * 0.18} F{new_filament_e_feedrate}\nG1 E{flush_length_2 * 0.02} F50\nG1 E{flush_length_2 * 0.18} F{new_filament_e_feedrate}\nG1 E{flush_length_2 * 0.02} F50\nG1 E{flush_length_2 * 0.18} F{new_filament_e_feedrate}\nG1 E{flush_length_2 * 0.02} F50\n; FLUSH_END\nG1 E-[new_retract_length_toolchange] F1800\nG1 E[new_retract_length_toolchange] F300\n{endif}\n\n{if flush_length_3 > 1}\n; FLUSH_START\nG1 E{flush_length_3 * 0.18} F{new_filament_e_feedrate}\nG1 E{flush_length_3 * 0.02} F50\nG1 E{flush_length_3 * 0.18} F{new_filament_e_feedrate}\nG1 E{flush_length_3 * 0.02} F50\nG1 E{flush_length_3 * 0.18} F{new_filament_e_feedrate}\nG1 E{flush_length_3 * 0.02} F50\nG1 E{flush_length_3 * 0.18} F{new_filament_e_feedrate}\nG1 E{flush_length_3 * 0.02} F50\nG1 E{flush_length_3 * 0.18} F{new_filament_e_feedrate}\nG1 E{flush_length_3 * 0.02} F50\n; FLUSH_END\nG1 E-[new_retract_length_toolchange] F1800\nG1 E[new_retract_length_toolchange] F300\n{endif}\n\n{if flush_length_4 > 1}\n; FLUSH_START\nG1 E{flush_length_4 * 0.18} F{new_filament_e_feedrate}\nG1 E{flush_length_4 * 0.02} F50\nG1 E{flush_length_4 * 0.18} F{new_filament_e_feedrate}\nG1 E{flush_length_4 * 0.02} F50\nG1 E{flush_length_4 * 0.18} F{new_filament_e_feedrate}\nG1 E{flush_length_4 * 0.02} F50\nG1 E{flush_length_4 * 0.18} F{new_filament_e_feedrate}\nG1 E{flush_length_4 * 0.02} F50\nG1 E{flush_length_4 * 0.18} F{new_filament_e_feedrate}\nG1 E{flush_length_4 * 0.02} F50\n; FLUSH_END\n{endif}\n; FLUSH_START\nM400\nM109 S[new_filament_temp]\nG1 E2 F{new_filament_e_feedrate} ;Compensate for filament spillage during waiting temperature\n; FLUSH_END\nM400\nG92 E0\nG1 E-[new_retract_length_toolchange] F1800\nM106 P1 S255\nM400 S3\nG1 X80 F15000\nG1 X60 F15000\nG1 X80 F15000\nG1 X60 F15000; shake to put down garbage\n\nG1 X70 F5000\nG1 X90 F3000\nG1 Y255 F4000\nG1 X100 F5000\nG1 Y265 F5000\nG1 X70 F10000\nG1 X100 F5000\nG1 X70 F10000\nG1 X100 F5000\nG1 X165 F15000; wipe and shake\nG1 Y256 ; move Y to aside, prevent collision\nM400\nG1 Z{max_layer_z + 3.0} F3000\n{if layer_z <= (initial_layer_print_height + 0.001)}\nM204 S[initial_layer_acceleration]\n{else}\nM204 S[default_acceleration]\n{endif}\n{else}\nG1 X[x_after_toolchange] Y[y_after_toolchange] Z[z_after_toolchange] F12000\n{endif}\nM621 S[next_extruder]A",
    "default_filament_profile": [
        "Bambu PLA Basic @BBL X1C"
    ],
    "default_print_profile": "0.20mm Standard @BBL X1C",
    "deretraction_speed": [
        "30"
    ],
    "extruder_clearance_height_to_lid": "90",
    "extruder_clearance_height_to_rod": "36",
    "extruder_clearance_radius": "57",
    "extruder_colour": [
        "#018001"
    ],
    "extruder_offset": [
        "0x2"
    ],
    "fan_kickstart": "0",
    "fan_speedup_overhangs": "1",
    "fan_speedup_time": "0",
    "from": "User",
    "gcode_flavor": "marlin",
    "host_type": "octoprint",
    "inherits": "",
    "is_custom_defined": "0",
    "layer_change_gcode": "; layer num/total_layer_count: {layer_num+1}/[total_layer_count]\nM622.1 S1 ; for prev firware, default turned on\nM1002 judge_flag timelapse_record_flag\nM622 J1\n{if timelapse_type == 0} ; timelapse without wipe tower\nM971 S11 C10 O0\n{elsif timelapse_type == 1} ; timelapse with wipe tower\nG92 E0\nG1 E-[retraction_length] F1800\nG17\nG2 Z{layer_z + 0.4} I0.86 J0.86 P1 F20000 ; spiral lift a little\nG1 X65 Y245 F20000 ; move to safe pos\nG17\nG2 Z{layer_z} I0.86 J0.86 P1 F20000\nG1 Y265 F3000\nM400 P300\nM971 S11 C10 O0\nG92 E0\nG1 E[retraction_length] F300\nG1 X100 F5000\nG1 Y255 F20000\n{endif}\nM623\n; update layer progress\nM73 L{layer_num+1}\nM991 S0 P{layer_num} ;notify layer change",
    "machine_end_gcode": ";===== date: 20230118 =====================\nM400 ; wait for buffer to clear\nG92 E0 ; zero the extruder\nG1 E-0.8 F1800 ; retract\nG1 Z{max_layer_z + 0.5} F900 ; lower z a little\nG1 X65 Y245 F12000 ; move to safe pos \nG1 Y265 F3000\nM991 S0 P-1 ;end smooth timelapse at safe pos\nM400 S2 ;wait for last picture to be taken\n\nG1 X65 Y245 F12000\nG1 Y265 F3000\nM140 S0 ; turn off bed\nM106 S0 ; turn off fan\nM106 P2 S0 ; turn off remote part cooling fan\nM106 P3 S0 ; turn off chamber cooling fan\n\nG1 X100 F12000 ; wipe\n; pull back filament to AMS\nM620 S255\nG1 X20 Y50 F12000\nG1 Y-3\nT255\nG1 X65 F12000\nG1 Y265\nG1 X100 F12000 ; wipe\nM621 S255\nM104 S0 ; turn off hotend\n\nM400 ; wait all motion done\nM17 S\nM17 Z0.4 ; lower z motor current to reduce impact if there is something in the bottom\n{if (max_layer_z + 100.0) < 250}\n    G1 Z{max_layer_z + 100.0} F600\n    G1 Z{max_layer_z +98.0}\n{else}\n    G1 Z250 F600\n    G1 Z248\n{endif}\nM400 P100\nM17 R ; restore z current\n\nG90\nG1 X128 Y250 F3600\n\nM220 S100  ; Reset feedrate magnitude\nM201.2 K1.0 ; Reset acc magnitude\nM73.2   R1.0 ;Reset left time magnitude\nM1002 set_gcode_claim_speed_level : 0\n\nM17 X0.8 Y0.8 Z0.5 ; lower motor current to 45% power\n\n",
    "machine_load_filament_time": "29",
    "machine_max_acceleration_e": [
        "5000",
        "5000"
    ],
    "machine_max_acceleration_extruding": [
        "20000",
        "20000"
    ],
    "machine_max_acceleration_retracting": [
        "5000",
        "5000"
    ],
    "machine_max_acceleration_travel": [
        "9000",
        "9000"
    ],
    "machine_max_acceleration_x": [
        "20000",
        "20000"
    ],
    "machine_max_acceleration_y": [
        "20000",
        "20000"
[... 106 lines cut ...]
        "300x300"
    ],
    "upward_compatible_machine": [
        "Bambu Lab P1P 0.4 nozzle",
        "Bambu Lab X1 0.4 nozzle",
        "Bambu Lab X1 Carbon 0.4 nozzle"
    ],
    "use_firmware_retraction": "0",
    "use_relative_e_distances": "1",
    "version": "0.0.0.0",
    "wipe": [
        "1"
    ],
    "wipe_distance": [
        "2"
    ],
    "z_hop": [
        "0.4"
    ],
    "z_hop_types": [
        "Spiral Lift"
    ]
}

```

SurgeHawk, 2023-07-23:
I was able to get the preset to show up by copying the whole BBL profile folder and the BBL.json in the profiles folder.

Faasteer, 2023-07-27:
> I was able to get the preset to show up by copying the whole BBL profile folder and the BBL.json in the profiles folder from bambu studio

I just tried it. The printer profile shows up (no picture), but when I want to add a printer, it just doesn't work. 
If it were possible to add this printer, that would be great. I like OrcaSlicer much better than BambuStudio.

SurgeHawk, 2023-07-27:
> I just tried it. The printer profile shows up (no picture), but when I want to add a printer, it just doesn't work. If it were possible to add this printer, that would be great. I like OrcaSlicer much better than BambuStudio.

It has been working great for me. Yes there isn't a picture. I had to put the check marks next to the profiles I wanted and then save. Restart orca and then the profiles show up in my drop down. I have been using it like this on the P1S for 2 days now.

Faasteer, 2023-07-27:
> It has been working great for me. Yes there isn't a picture. I had to put the check marks next to the profiles I wanted and then save. Restart orca and then the profiles show up in my drop down. I have been using it like this on the P1S for 2 days now.

You're the best! The reboot, that was the one thing I didn't think of. I'm gonna go test it now, but it looks good. Thank you. 👍

nirurin, 2023-07-27:
Does doing this mean the printer/filament/print profiles saved on bambu studio for the P1S auto-import? Or will that require waiting for orcaslicer to update to include the p1s properly?

SurgeHawk, 2023-07-27:
> Does doing this mean the printer/filament/print profiles saved on bambu studio for the P1S auto-import? Or will that require waiting for orcaslicer to update to include the p1s properly?

It currently uses all the profiles for X1C. I have been using the X1C profiles from Bambu (Imported when copying from Orca) and have not had any issues.

Faasteer, 2023-08-02:
> I was able to get the preset to show up by copying the whole BBL profile folder and the BBL.json in the profiles folder from bambu studio

After a few days I can say it works great. Except for one thing, you can't select the filament type in AMS. There is no entry. Fortunately, I can do it in BambuStudio and then use Orca, which is a much better slicer for me.

Faasteer, 2023-08-02:
I think the problem is only that the P1S is partly a brand new model that is not yet officially supported. Even Bambu Studio only had it in beta until today. With the trick recommended by @SurgeHawk everything works for me, including AMS. The only thing that doesn't work is setting up new material in a specific slot. 

No big deal, it can be worked around for now. And I believe they will fix it soon.

closed 2023-11-22 by github-actions[bot]