OrcaSlicer/OrcaSlicer #1581: FR: 2in1out hotend

stenc55, 2023-07-23:
I have modified Sapphire SP5 with with 2 extruders that share Thaici 2in1out hotend. Super easy to set in PS or SS (Single extruder multi material option), but in Orca there is no option to set parameters for switching extruders/ forming tips/ changing filaments. 
I guess something could be coded in switching filaments section of profile, but I can't write the code. 
I have Switch filament macro written in Klipper, but it is static. Printhead does not move while changing filament. I could make it move while purging filament, but I can't insert wipe tower position in macro or vice versa, I can't insert purging parameters into wipe tower generated in slicer?

I would like setup similar to PrusaSlicer. Cooling tube parameters, filament ramming and forming tips parameters and wipe tower parameters. 
Until I solve this problem unfortunately Orca Slicer is not usable for me.
[image]
[image]
[image]

TexZeTech, 2023-08-03:
Question for you. Have you tried 

[image]
and havinging custom gcode set up in the second filament? Something like `extruderchange 2` or the like to call the macro in klipper? I think you can get the functionality you want in Orcaslicer.

stenc55, 2023-08-04:
I have 2 filaments set up.
Problem is, Orca slicer only know how to use AMS configuration, I believe. When I try filament change, it will stop with nozzle still in print, perform filament retract (with that forming tip) and only then it will move to purge tower, swap extruders and load new filament. 
While forming tip it will push some old filament out of nozzle and then move filament in/out several times to make a tip.
I can't set it to move to purge tower and then perform everything necessary for filament swap.
in picture, T0 code should be in place of ;CP TOOLCHANGE WIPE
[image]

TexZeTech, 2023-08-04:
So playing around a little more. My first suggestion was WAY off the mark. 

I cannot test it out besides just looking at the gcode generated.

Here is what I have poked about and I think would work for what you need. 

Again this would be dependent on your macro setup. 

[image]

[image]

I legitimately hope this solves your issue.

stenc55, 2023-08-04:
Thank you.
I was thinking same way, but I still don't know how to cancel T0/T1 command that is inserted after filament start gcode. Or at least comment it out.  Can commenting out a line be done?

TexZeTech, 2023-08-04:
I found something that might work, it goes without saying you will have to mod the macro to suit your system.

It would help if you upload your config(s) as well.

https://klipper.discourse.group/t/switching-between-extruders/2020/19

https://github.com/ld3300/Klipper_Configs/blob/master/configs/extruders.cfg

I don't think the T0, T1, etc is an issue personally as I can use a macro to use that for more control personally.

I did find though that if I try to use the support/raft interface as a separate filament selection I didn't get any T# .... that has to be a bug.

stenc55, 2023-08-05:
Thank you again. I will play with this when I have time, right now there are biblical floods in my country.

TexZeTech, 2023-08-05:
> Thank you again. I will play with this when I have time, right now there are biblical floods in my country.

Ooooof well stay safe and no problem.

stenc55, 2023-09-26:
Implemented with one small issue, described in #2235

closed 2023-09-26 by stenc55