OrcaSlicer/OrcaSlicer #1593: Can't load my profile after startup

Bumblebiber, 2023-07-25:
I've freshly installed OrcaSlicer and set up a profile.

But even before that, I've created a shortcut of the orca-slicer.exe with -datadir "Z:\software-tools\Slicer\PrusaSlicer 2.6\Settings" so I can access my profiles easily. This works fine for printer settings and filament settings. They are being loaded on startup.

But for Process I can only choose from standard profiles (even though my created profile is still in the User folder). 

I've tried to overwrite a standard profile, but this gets overwritten on every startup.

Bonaparte1446, 2023-07-26:
hi, I had also the same bug, logout an login and everything should return ok.

Bumblebiber, 2023-07-26:
> hi, I had also the same bug, logout an login and everything should return ok.

Hi, log out from where? I haven't even made an account.

closed 2023-11-22 by github-actions[bot]