OrcaSlicer/OrcaSlicer #1595: Print flow ratio of multi-part object set incorrectly

Mqrius, 2023-07-25:
**Describe the bug**
If you have an object made of multiple parts, the slicer uses the print_flow_ratio of the first part for the entire object.

**3mf File for This Bug**
[Flow Ratio Bug Report.zip](https://github.com/SoftFever/OrcaSlicer/files/12165255/Flow.Ratio.Bug.Report.zip)

**To Reproduce**
In a new file, add a primitive cube. To that cube, add a second part, eg another cube. For the first part, set the print_flow_ratio (Object>Quality>Advanced>Flow Ratio) to 0.8, and for the second part set the print_flow_ratio to 1.2. Slice the plate, and set color scheme to Flow. Observe that the first part's flow rate is used for the entire object. Even if you set a third flow ratio for the parent object, this is ignored, and the first part's flow ratio is used.

**Expected behavior**
Expected flow rate of the first part should be lower than the second part of the object. If there are any parts without an adjusted flow ratio, they should use the parent object's flow ratio.

OrcaSlicer Version 1.6.3

SoftFever, 2023-07-30:
Hi @Mqrius 

Can confirm it's a bug. 
Given the impact of this issue and the cost of testing, it has low priority to fix though.

Mqrius, 2023-07-30:
Low impact is fair enough, but cost of testing is very low — the incorrect flow rate shows up in the slicer preview, so there's no need to print anything. You can see the difference by viewing the Flow and/or the Speed color scheme.

SoftFever, 2023-07-30:
> Low impact is fair enough, but cost of testing is very low — the incorrect flow rate shows up in the slicer preview, so there's no need to print anything. You can see the difference by viewing the Flow and/or the Speed color scheme. I figured it was useful to report it even with low bugs since it might be linked to some systemic other bugs that haven't been discovered yet.

Hehe, the testing cost mainly arises because the dev have to figure out and test as many scenarios as possible to ensure that a code change won't break or regress existing functionalities. So far, this is the most time-consuming part of the whole development.

> I figured it was useful to report it

Absolutely, I really appreciate it!

Mqrius, 2023-07-30:
Ah in that sense, yeah that makes sense :D

closed 2023-11-22 by github-actions[bot]