OrcaSlicer/OrcaSlicer #1605: Adding the P1S as a printer config / option

nirurin, 2023-07-27:
Doesn't really need any more details. P1S as a printer option.

Error-106, 2023-07-27:
I found the configuration file containing P1S from AppData\Roaming\BambuStudio\ota
You just need to copy it to OrcaSlicer\resources\profiles replacement
[ota.zip](https://github.com/SoftFever/OrcaSlicer/files/12182978/ota.zip)

nirurin, 2023-07-27:
> I found the configuration file containing P1S from AppData\Roaming\BambuStudio\ota You just need to copy it to OrcaSlicer\resources\profiles replacement [ota.zip](https://github.com/SoftFever/OrcaSlicer/files/12182978/ota.zip)

Does this mean that it will auto grab the custom profiles for p1s from the server that I have made on bambu studio? The same as it would for p1p and x1c profiles

Error-106, 2023-07-28:
Bambu Studio will attempt OTA at startup
OrcaSlicer should not attempt to update
You can copy the Bambu Studio OTA configuration file to OrcaSlicer, if only temporarily

Brian-Jackson, 2023-07-30:
What does if only temporarily mean?  Since it is an OTA update for Bambu does that mean in order to update Orca we would need to do this again to get a new version with any changes that Bambu makes?

Error-106, 2023-07-30:
I'm not sure OrcaSlicer has a license to use BamBu Studio's OTA profiles
You can use it temporarily and wait for OrcaSlicer to update their configuration file

closed 2023-11-22 by github-actions[bot]