OrcaSlicer/OrcaSlicer #1612: Take accel_to_decel into account for print time estimate

bonjorno7, 2023-07-27:
**Is your feature request related to a problem? Please describe.**
My prints take on average 20-30% longer than the slicer estimate.
I can get an accurate result from klipper estimator, but it's a hassle (because I get an error when using it as post-processing script so I have to run it manually on the exported gcode file) and doesn't actually affect the UI inside the slicer.
If I halve all my acceleration values and set jerk to `sqrt(2) * square corner velocity` I get a pretty accurate estimate, but then that of course affects the gcode in ways I don't want.

**Describe the solution you'd like**
Make the time estimate math inside the slicer take accel_to_decel into account, for example by multiplying acceleration by the accel_to_decel percentage (it's not exactly accurate but better than nothing).
Replace Jerk with Square Corner Velocity when using Klipper, because it's not quite the same thing.

**Describe alternatives you've considered**
Only using acceleration and jerk settings for estimates, as described in issue #867 (except there it's about machine limits and here it's about print profile settings) would allow me to just set these values to something that gives an accurate enough estimate, without affecting the actual gcode.
Though I think this only makes sense if you're using the same acceleration and jerk for every line type.

**Additional context**
I could add a screenshot showing a 6 hour slicer estimate next to an 8 hour real print time, but I think you get the idea.

SoftFever, 2023-08-01:
@bonjorno7 

> If I halve all my acceleration values and set jerk to sqrt(2) * square corner velocity I get a pretty accurate estimate

Wow, this works like a charm. 
Could you shed some light on where to find more details about the comparison between Marlin's jerk and Klipper's SCV?

The klipper_estimator is supported by Orca.
You probably need to use `--config_moonraker_url` instead of `--config_file` for post-process mode.
Something like this:
`c:/Apps/klipper_estimator.exe --config_moonraker_url http://192.168.1.3:10088 post-process;`
[image]

bonjorno7, 2023-08-01:
I think I got the square root from this reddit comment: https://www.reddit.com/r/klippers/comments/vse7ir/comment/if1i1ie/
Though it may not be accurate for angles other than 90 degrees, because the math is different (I don't remember my source for this unfortunately).

There's also the formula for junction deviation which is `square_corner_velocity = sqrt(junction_deviation * max_accel * (sqrt(2) + 1))`, from this github issue: https://github.com/Klipper3d/klipper/issues/468

As for klipper estimator I have tried it with moonraker url (unsuccessfully), though my url didn't have a port.
I think it did work when running from CMD, but I got the same error in slicer; will have to experiment more.

SoftFever, 2023-08-05:
> I think I got the square root from this reddit comment: https://www.reddit.com/r/klippers/comments/vse7ir/comment/if1i1ie/ Though it may not be accurate for angles other than 90 degrees, because the math is different (I don't remember my source for this unfortunately).
> 
> There's also the formula for junction deviation which is `square_corner_velocity = sqrt(junction_deviation * max_accel * (sqrt(2) + 1))`, from this github issue: [Klipper3d/klipper#468](https://github.com/Klipper3d/klipper/issues/468) And this doc is worth looking at: https://github.com/Klipper3d/klipper/blob/master/docs/Kinematics.md

He didn't explain how he get the number.

Interestingly, when I used the equation for junction_deviation and SCV that you mentioned for Klipper, and the equation found [here ](https://www.3dmakerengineering.com/blogs/3d-printing/velocity-acceleration-jerk-and-junction-deviation) for Marlin
[image]

I get `jerk= sqrt(2.5) * SCV`
However, neither sqrt(2.5) nor sqrt(2) worked well in complex models. 
The difference between these two algorithms cannot be simplified as a linear factor.

Meanwhile, klipper_estimator works perfectly. 
Instead of reinventing the wheel, I would simply recommend using it.

bonjorno7, 2023-08-05:
It does work, but being a post-processing script means it doesn't actually update the UI in the slicer, which is not ideal.
Tighter integration with klipper estimator would solve that I suppose.

SoftFever, 2023-08-05:
agreed.

SoftFever, 2023-08-05:
Close it as not planned.

closed 2023-08-05 by SoftFever