OrcaSlicer/OrcaSlicer #1613: Feature request: Flow rate calibration improvement with useful default squares

tastyratz, 2023-07-28:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]
The current flow rate calibration test is not really useful in the existing state. 
Pass 1 involves significantly large step variations that are well outside the range of normal printers. a 40% range on flow rate is not applicable to anyone unless using violently worn nozzles. I cannot picture a scenario where you might need to calibrate 120%-80% flow.
Pass 2 also is ONLY subtractive which means choosing a flow rate that is far too high or hand editing each square.

**Describe the solution you'd like**
Use the [Ellis tuning guide](https://ellis3dp.com/Print-Tuning-Guide/articles/extrusion_multiplier.html) examples instead of 2% flow steps on pass 1 and 0.5% flow steps on pass 2. Make them both additive and subtractive from the current flow setting.

**Describe alternatives you've considered**
Doing this manually by hand gets the job done but it defeats the purpose of generating calibration squares

jeremytodd1, 2023-07-28:
I also agree that the Pass 1 flow test is basically pointless. I always skip it and go straight for flow test 2.

Personally, I've never had filament be above 100% flow. Saying that, I think it'd be nice if the flow test was configurable like most of the other tests. Let me put in what starting flow rate I want to start with and then let me put in what step size I want to do.

Also, the current tests only have 1 layer of infill. It'd be nice if we could bump that up to 2 or 3 layers of infill. This will help ensure that any first layer squish is definitely mitigated. Whenever I do flow tests I typically scale up the Z a little bit to give it more infill layers but it would nice to not need to manually do this.

tastyratz, 2023-07-28:
Thinking about this there is ONLY a SINGLE instance I can think of where flow rate could ever possibly need such great variance and it's the Colorfabb Varioshore filament but that's priced out of hobbyists' regular use for some reason (must be expensive baking soda) and impossible to actually find at anybody's regular distributors. I wouldn't say it's worth the exception.

Fully configurable range steps might mean new models with embossed numbers won't match (not that it does today) and would probably make more sense being sequentially numbered since the slicer won't actually be generating the STL's.

SoftFever, 2023-07-30:
The testing is designed as generically as possible. The current flow rate calibration's design goal is to ensure that if the user follows the steps (2 passes), they will be able to get a decent result. We can't simply assume that all extruders are perfectly calibrated in the real world of 3D printing.

tastyratz, 2023-07-30:
I would agree that we wouldn't assume this, but, a 40% range?

That's invalidly defined nozzle, uncalibrated rotation distance, severe misconfiguration territory. 
Few filaments will ever require greater than 100%. Poorly calibrated machines might dig down to 90%. That's what I would call a likely poorly calibrated machine.

Have you ever heard of someone running a warranted 80% or 120%? Even 85%-115%?
I might venture to say 98% of slicer users would fall into a 92-99% extrusion rate range. Would you disagree?

nirurin, 2023-07-31:
The range in this is definitely ludicrous. A machine that is so badly calibrated as to need a flow multiplier of 20% lower or higher just wouldn't print under normal circumstances. Anything beyond 10% (and arguably less that that) should prosude a result of "error: check your esteps and then try again" with a link to the method of editing esteps for the firmware they are using. Allowing the user to mask a major printer issue by using extreme multiple values is frankly unhelpful, and likely to cause that user significant issues with their printer. 

I had a similar issue with the other calibration test in orca (the pressure advance test). The orca one was antiquated and useless, and there was a better variation out there (again, from Ellis). Unfortunately the dev refused to consider updating as he thought the existing test was 'good enough' (it wasn't). Fortunately an amazing community member took up the job and ported the proper calibration test into orca for us.

I think our only chance may be if another community member edited this calibration to have more options available in the drop downs, or made it customisable. If it hasn't happened by the time I hand in my thesis, I'll take a stab at it myself (I need to avoid the distraction right now lol)

SoftFever, 2023-08-01:
> I would agree that we wouldn't assume this, but, a 40% range?
>
> That's invalidly defined nozzle, uncalibrated rotation distance, severe misconfiguration territory. Few filaments will ever require [...]

It's +/- 20%, not 40%. It's important to know the difference between these concepts when dealing with tolerance.

The flow rate can be affected by multiple factors, such as extruder calibration, filament characteristics, and filament diameter. Moreover, we need to consider the starting 'flow ratio' value in the filament setting for calibration. Many beginning users simply copy these values from random places.

If someone knows their printer is calibrated and has a rough idea of the specific filament's flow rate range, they can skip some steps. For example, I set the starting flow ratio to 0.94 and proceed with only pass 2 when calibrating ABS on a machine that I'm familiar with.

However, you and I don't represent the majority of the user base, not at all. For experienced users, there are many ways to shorten the testing process.

The two-pass approach designed in Orca is meant to cover most scenarios. Any user, experienced or not, can use the two-pass flow rate calibration to calibrate any unknown filament with just two testing prints. That's the goal of this calibration approach.

Changing the first pass's range from +/- 20% to +/- 10% doesn't save anything from a UX point of view; we still need to print two test prints to find the optimal flow rate. So, why bother?

nirurin, 2023-08-01:
> It's +/- 20%, not 40%. It's important to know the difference between these concepts when dealing with tolerance.

I'm pretty sure he referred to it as a "40% range". Going from -20 to +20 is a range of 40. 

It's important to know the difference between these concepts when dealing with tolerance.

SoftFever, 2023-08-01:
It seems that this discussion is heading in a direction that will not benefit anyone. 
I'm closing this thread now. 
Thank you.

closed 2023-08-01 by SoftFever

tastyratz, 2023-08-02:
While this thread wasn't as productive as I hoped, I did want to follow up here. I was in fact referencing a range of 40% in total. I think if we're setting that as a reasonable expectation you're both not helping advanced users with tuned printers and encouraging/enabling improperly configured printers that are going to instead create downstream support cases.

I know it's not a priority or shared opinion here with you SoftFever and the userbase for Orca but for anyone willing to take this on Andrew Ellis has made a number of test cubes available with respective numbering on them. Since the Ellis tuning guide is prettymuch the gold standard of reference I thought I'd drop a link to the cube prints:

https://github.com/AndrewEllis93/Print-Tuning-Guide/tree/main/test_prints/extrusion_multiplier_cubes/labeled

This way maybe someone can create a calibration pass useful to the majority of users with them should they have the capabilities.

Aasikki, 2024-06-11:
Sad to see this closed. I agree that the current "settings" are good defaults, to cover every possible situation. But don't all other calibrations have user configurable settings? Why now have that for flow calibration as well? Having user configurable range and steps would make it much more useful imo.

tastyratz, 2024-06-11:
> Sad to see this closed. I agree that the current "settings" are good defaults, to cover every possible situation. But don't all other calibrations have user configurable settings? Why now have that [...]

I don't know that user configurable is as easily done since the calibration squares are going to be preselected models. I do still firmly stand behind my belief that the inbuilt range is twice as big as it should reasonably be for almost the entire userbase and encourages users to set inappropriate flow rates that lead to weird results instead of fixing their printers.
Anyone who has some kind of strange use case validating the range should be using custom test squares.

In my own usage, I avoid this by having my own project full of appropriately ranged test squares and encourage others to do the same.