OrcaSlicer/OrcaSlicer #1614: VFR Max calibration regression in nightlies

psiberfunk, 2023-07-28:
**Describe the bug**
The Max VFR calibration in the nightly builds has suddenly started showing two lines instead of one.  I'm not able to pinpoint which commit changed this, but it breaks the max VFR test and should be fixed.

**To Reproduce**
Run the Max VFR calibration in the slicer, observe the preview

**Expected behavior**
There should be 1 wall, not 2.  Two walls will effectively double the test time and may compromise test results.  No need to waste this plastic.

**Screenshots**
[image]

**Desktop (please complete the following information):**
Windows 11
1.6.4-dev , pulled from this github action:  https://github.com/SoftFever/OrcaSlicer/actions/runs/5682999165

psiberfunk, 2023-07-28:
@SoftFever I found this testing the nightlies, not sure precisely when it was introduced.

psiberfunk, 2023-07-28:
To clarify this is definitely not in 1.6.3 and was not in some earlier 1.6.4 nightlies .. so it got introduced in the last few weeks I think

psiberfunk, 2023-07-28:
Update based on nightlies testing:  This would appear to be fixed as of https://github.com/SoftFever/OrcaSlicer/actions/runs/5693025113 at least ?

psiberfunk, 2023-07-30:
Tested in https://github.com/SoftFever/OrcaSlicer/actions/runs/5706149196 , which seems to have fixed the issue.  Feel free to close this issue as resolved.

SoftFever, 2023-07-30:
Thanks for testing and reporting it 👍

closed 2023-07-30 by SoftFever