OrcaSlicer/OrcaSlicer #1616: Chevron PA Pattern fails to hit max flow , outer wall speeds changed AND not respected

psiberfunk, 2023-07-28:
**Describe the bug**
This is a bit of a weird possible combination glitch, discovered when I realized the flow for the Ellis/Chevron PA pattern wasn't right by default.  There are two major issues here:

1. The defaults for the new PA pattern (Chevron style) for some reason changes the outer wall speed to 100 by default.  I'm not sure why this is done but it causes the pattern not to hit the max flow if VFR>~8 mm^3/sec on a standard 0.4 nozzle.  This is not a good default test setting as it is very common to go over this.  I'm not sure why the profile's outer wall speed should be altered for this test?

2.  The outer wall speed SHOULDNT actually break the test, but because somehow the slicer is applying the outer wall speed instead of the internal infill speed (which the slicer says the lines are on inspection), the test is broken and higher VFRs don't get tested right.

**To Reproduce**
Using this Win64 build:  https://github.com/SoftFever/OrcaSlicer/actions/runs/5693025113 :

1. Pick a filament profile with VFR=15, for example
2. setup the PA Pattern (chevron test) with defaults.
3. Slice. 
4. Observe that the max flow with a 0.4 Nozzle and bambu defaults shows a max flow of ~ 7.98 or so, as shown below in screenshots.
5. Switch to Line type mode.
6. Observe that somehow the outer wall speed is applying to lines marked as internal infill (purple).  This seems to be a bug too.

**Expected behavior**
The default test should not limit the max wall speed.  
Outer wall speed should not be applying to internal infill lines.

**Screenshots**
Max flow of ~7.98:

[image]

Internal infill lines are getting outer wall speed limited:
[image]

psiberfunk, 2023-07-30:
Additional comment here:  I've also just discovered that the chevron test is not respecting the max volumetric flow limit settings in the filament setting.  To see this, set the max VFR for the current filament = 1, and re-slice... you will see that the flow view shows you that the flow far exceeds 1 !

@SoftFever  do you want a separate bug filed for this bad behavior or should it be handled in this ticket ?

psiberfunk, 2023-11-06:
@SoftFever this is still real AFAIK

psiberfunk, 2024-02-12:
This isn't stale, it's still a real issue.

psiberfunk, 2024-05-13:
still not stale

psiberfunk, 2024-08-12:
I don’t believe this is stale @SoftFever .. should it get marked as won’t do , or is it still considered ?

MxBrnr, 2024-10-25:
I can confirm that in both 2.2.0-beta2 and in 2.2.0-RC, the PA Pattern test is still ignoring the filament's max vol. flow settings, and additionally ignoring line widths of less then 0.42mm (when using a 0.4mm nozzle), and ignoring travel acceleration. These two latter issues are not problems in the original Ellis3DP PA tuning tool.

psiberfunk, 2024-10-25:
yes, this has long been an annoyance I wish would be remedied :)

MxBrnr, 2024-10-25:
Also just discovered and confirmed with testing that the accels being used when actually printing the calibration are not following a standardized order. The first layer is printing using the first layer accel and speed (as it should be). The the second layer is printing using the outer wall speed (as it should) and the "normal printing" accel (odd, but ok), but then any subsequent layers are printing using the outer wall speed with the "Internal solid infill" accel (that's definitely not right). The accels noted here are also the ones being used for any travel moves during their respective layers. This is with Orca 2.2.0-RC

psiberfunk, 2025-01-24:
I don’t think this is fixed still ?