OrcaSlicer/OrcaSlicer #1617: Monotonic Infill

Pamapama, 2023-07-28:
*Is your feature request related to a problem? Please describe.*
I print some parts (sliced with SuperSlicer) with 100% Infill, monotonic sparse pattern at 0° angle for transparent petg parts. SuperSlicer does have this feature. 

*Describe the solution you'd like*
Please add monotonic Infill pattern

*Describe alternatives you've considered*
---

*Additional context*
Transparent petg print are much better with 0° fill angle increment and monotonic Infill pattern

Noisyfox, 2023-07-30:
I think aligned rectilinear should work.

zoroseerus, 2023-08-07:
Second this. Bambu Studio just got around to adding this in 1.7 and I coincidentally also ran into a print where I'd like to do the same.

Aside, is it better to simply comment here about the request, or would creating more requests for the feature be better? I haven't actually posted on GitHub before and don't see any sort of upvote or visibility increasing mechanic

Eldenroot, 2023-11-06:
Plus oné:)

igiannakas, 2023-12-26:
This has been implemented

closed 2024-04-02 by github-actions[bot]