OrcaSlicer/OrcaSlicer #1619: AppImage Launch Issue - JavaScript: Unsupported result type

Dillacorn, 2023-07-29:
Similar issue on BambuStudio
https://github.com/bambulab/BambuStudio/issues/1782

terminal command:
./OrcaSlicer_ubu64.AppImage 

output: (launch fault)
[2023-07-28 20:11:37.210602] [0x00007f132d558a40] [trace]   Initializing StaticPrintConfigs
add font of HarmonyOS_Sans_SC_Bold returns 1
add font of HarmonyOS_Sans_SC_Regular returns 1
Segmentation fault

expected output:
The application should launch

OS Build:
Debian 12 Bookworm + GNOME 43.6

Windowing System:
Wayland

When I use strace the application launches.. I can't read the errors.. let me know if you need something there.

Warning:
Error running JavaScript: Unsupported result type

[image]

Dillacorn, 2023-07-29:
Just tried Xorg X11 instead of Wayland session and experienced the same issue.

Sorry if this is a trivial error and I'm just not seeing the big picture as to why this is occurring on my instance.

emelbardis, 2023-08-02:
@WayneManion 
"errors are gone in the BambuStudio 1.7 betas"
Using what flavor Linux? I run Fedora and haven't been able to get a working appimage yet with original BS or OS. 
I really wish someone who better knew their way around appimages would contribute to the project :)

WayneManion, 2023-08-02:
I run Arch.

Dillacorn, 2023-08-02:
I'm now running PrusaSlicer till Orca is fixed. It has a profile for the Ender 3 Pro. tweaked some of the settings and I can confirm it's a little harder to navigate compared to Orca but will work just fine till Orca receives a patch.

Dillacorn, 2023-08-09:
Tried "1.6.4 Beta 2" release and getting the same bug upon attempting to run the application.

[2023-08-09 08:49:39.788232] [0x00007f1ba73ffa40] [trace]   Initializing StaticPrintConfigs
add font of HarmonyOS_Sans_SC_Bold returns 1
add font of HarmonyOS_Sans_SC_Regular returns 1
add font of NotoSansKR-Regular returns 1
add font of NotoSansKR-Bold returns 1
Segmentation fault

The "NotoSansKR" are new lines

georgewoodall82, 2023-08-24:
still broke for me on version 1.6.6

georgewoodall82, 2023-08-25:
I may have found a fix, I re installed arch and for some reason it worked, and then i installed a custom gtk theme and it broke.
If you are using a custom GTK theme, maybe try changing it to Adwaita, which works for me.
I just used `GTK_THEME=Adwaita ./OrcaSlicer*.AppImage`

Dillacorn, 2023-08-25:
Ah so possibly non-default GTK themes break the appimage.. I was on Gnome and now on KDE Plasma so maybe it will work for me now. I'll have to test it later. Given the errors I was seeing this makes sense.. I'll still probably stick with Prusa or Bambu if this is the case though because they don't have this issue and I prefer my nordish theme. I hope this issue can be fully resolved.

RAP2022, 2023-08-26:
Hi,
The JavaScript Dialog Box warning issue is "fixed" or hidden  when I apply the below two commtis:
https://github.com/bambulab/BambuStudio/commit/0699ee2da33408691e4299355e7a3a4596727c0c
https://github.com/bambulab/BambuStudio/commit/ca0847d0ac2f4c16f9d0f442bf7a8dad3767a13c

outpoints, 2023-08-27:
Thanks @RAP2022!

closed 2023-12-04 by github-actions[bot]