OrcaSlicer/OrcaSlicer #1621: 关于切片层高的问题。。。

slayercjc, 2023-07-29:
我的层高设置的0.2，但是切片完以后看切片里面的顶层第一层是0.39层高，也就是第147层，147层再往上都正常比如148就是0.2层高，不知道为什么会出现超过我设置的0.2层高，不知是我设置错误，还是软件的BUG，求解答。
[image]
[image]

SoftFever, 2023-07-30:
> 我的层高设置的0.2，但是切片完以后看切片里面的顶层第一层是0.39层高，也就是第147层，147层再往上都正常比如148就是0.2层高，不知道为什么会出现超过我设置的0.2层高，不知是我设置错误，还是软件的BUG，求解答。
> [image]
> [image]

搭桥时如果勾选了thick bridge的话，搭桥的层高会按喷嘴的直径算。
因为是搭桥，所以不会造成问题

Noisyfox, 2023-07-30:
不选thick bridge的话就是Bridge flow参数的问题，因为在这个情况下对于这一层搭桥的层高是 sqrt(bridge flow) * 喷嘴直径，所以如果要一样的层高那么bridge flow设置成0.25即可

slayercjc, 2023-07-30:
> > 我的层高设置的0.2，但是切片完以后看切片里面的顶层第一层是0.39层高，也就是第147层，147层再往上都正常比如148就是0.2层高，不知道为什么会出现超过我设置的0.2层高，不知是我设置错误，还是软件的BUG，求解答。
> > [image]
> > [image]
> 
> 搭桥时如果勾选了thick bridge的话，搭桥的层高会按喷嘴的直径算。 因为是搭桥，所以不会造成问题

thick bridge 是厚桥的意思吗？我没有选择这个啊

slayercjc, 2023-07-30:
> 不选thick bridge的话就是Bridge flow参数的问题，因为在这个情况下对于这一层搭桥的层高是 sqrt(bridge flow) * 喷嘴直径，所以如果要一样的层高那么bridge flow设置成0.25即可。 不过确实不管这个也不会有问题。

bridge flow是搭桥流量的意思吗？我设置的是0.95，设置成0.25是不是就没啥流量了。。。

Noisyfox, 2023-07-30:
那肯定，这个的主要的原因是因为搭桥的话，它下层是悬空的，所以没有办法在正常挤出量的情况下获得比喷嘴直径更小的层高。 其它层能用0.4mm的喷嘴获得比如0.2mm的层高靠的是下层的挤压，而桥因为是悬空的，所以料从你喷嘴出来是多粗那么它的层高就是多高，除非你在同样的喷嘴移动速度下降低流量强行把料拉细

SoftFever, 2023-07-30:
同意楼上 👍

closed 2023-07-30 by SoftFever

slayercjc, 2023-08-02:
> 那肯定，这个的主要的原因是因为搭桥的话，它下层是悬空的，所以没有办法在正常挤出量的情况下获得比喷嘴直径更小的层高。 其它层能用0.4mm的喷嘴获得比如0.2mm的层高靠的是下层的挤压，而桥因为是悬空的，所以料从你喷嘴出来是多粗那么它的层高就是多高，除非你在同样的喷嘴移动速度下降低流量强行把料拉细。 具体怎么计算的，可以看https://manual.slic3r.org/advanced/flow-math
> 
> [image]

谢谢大佬解答。。。