OrcaSlicer/OrcaSlicer #1628: Combing Not in Skin

LoboThirteen, 2023-07-29:
Is it possible to implement a function known from Cura Slicer?  "Combing Not in Skin".  I print without Z-Hop because I don't actually need it with my RatRig.  Unfortunately, this often messes up the top player, for example the inner bottom of a cylinder.

Attached is a link where the combining function is described.

https://all3dp.com/2/combing-mode-cura-simply-explained/

jeremytodd1, 2023-07-29:
This feature will be in the next release. The work on it just wrapped up on this a few days ago, actually (in pull request 1562)

You can already use the feature if you download the latest nightly.  It's called Z Lift Enforcement and it does have an option to only be used on top layers.

[image]

Please close this issue as it is no longer needed.

LoboThirteen, 2023-07-29:
Cool, thank you for the quick reply!! 😊

LoboThirteen, 2023-07-29:
Problem solved...

closed 2023-07-29 by LoboThirteen