OrcaSlicer/OrcaSlicer #1634: [BUG] Settings don't work unless restarting Orca first, non-Marlin Gcodes

rondlh, 2023-07-30:
I'm a newbie user, and just gave Orca 1.63 (Windows) a try, I hope this feedback is helpful. I noticed some issues...

I setup the printer (Marlin(legacy)), material and bed temps for the high temp plate, selected the high temp plate, but the produced Gcode did not contain any M140 or M190. After restarting Orca M190 appeared in the Gcode.

Also I found 2 commands which are not supported by Marlin:
1.
M981 S1 P20000 ;open spaghetti detector

2.
; open powerlost recovery
M1003 S1

**To Reproduce**
Start Orca in a environment where Orca was never installed, set the bed temps and select the plate, and check the Gcode file generated for M190, M981 and M1003.

**Printer model**
Custom based on MKS Monster 8 V1.0 and BTT TFT35 V3.0

**Desktop**
Windows 10 x64
Orca 1.63

SoftFever, 2023-07-30:
Hi @rondlh 
I guess you use a Bambu printer as a template.
Please start with a Generic Marlin Printer template:
[image]

rondlh, 2023-07-30:
> Hi @rondlh
> I guess you use a Bambu printer as a template.
> Please start with a Generic Marlin Printer template:

I did use the Generic Marlin Printer I believe, I'm not running Klipper. Let me try again...

rondlh, 2023-07-30:
Probably wrong template used when setting up the config the first time, no follow up needed.

closed 2023-07-30 by rondlh