OrcaSlicer/OrcaSlicer #1635: [BUG] Process settings disappear if you save and load the project.

FillPavS, 2023-07-30:
If I change any settings like number of walls, speed, number of bottom layers, top and any others, then save the project and load it, then all settings will be lost and the ones saved by default in the process profile will be used.

**3mf File for This Bug**
no model needed to reproduce or any

**To Reproduce**
1. Run the program
2. Change any parameters in the process - the number of walls, bottom layers, top, speed. Any
3. Save the project
4. Open this project (do not save changes to the process profile)
5. Switch tabs - Quality, Strength, Speed

You will see that your changes are gone and the settings saved in the process profile are being used.

**Printer model**
Bambu Lab X1C

**Desktop (please complete the following information):**
 - OS: Windows 10
 - Version last from Actions https://github.com/SoftFever/OrcaSlicer/actions/runs/5704604879
 - The version I'm using without this error https://github.com/SoftFever/OrcaSlicer/actions/runs/5578530790

SoftFever, 2023-07-30:
@FillPavS 
Thanks for testing and reporting it.
This should have been fixed in the last nightly build: https://github.com/SoftFever/OrcaSlicer/actions/runs/5706900746

FillPavS, 2023-07-31:
@SoftFever Now everything is fine. Thank you

closed 2023-07-31 by FillPavS