OrcaSlicer/OrcaSlicer #1637: Pattern selection for Bridge infill

bengone1, 2023-07-30:
The Bridge Infill pattern is fixed which works for most objects but when printing a spoked or hub pattern the linear pattern will end up running lengthwise along some of the unsupported areas. You can rotate it but that just changes the spoke overhang left unsupported. 

**Describe the solution you'd like**
The ability to disable bridging, select concentric, or Archimedean patterns

This seems to be unresolved in the PrusaSlicer as there are quite a few posts online with no solution.

[image]

SoftFever, 2023-07-31:
Interesting.
I experimented with it a few months back.
I didn't add it eventually as I planned to do more experiments. And also was not sure how many users atually need this.
Could you share your model?
[image]

bengone1, 2023-07-31:
Sure thing, https://www.printables.com/model/450634-nintendo-switch-game-holder-updated-for-correct-st

closed 2023-11-21 by github-actions[bot]