OrcaSlicer/OrcaSlicer #1639: Ender 2 Pro

Datt-Flo, 2023-07-30:
Is there any possibility that the Ender 2 Pro will be included in the Orca Slicer ?

closed 2023-11-21 by github-actions[bot]

thunder175, 2024-01-13:
Any chance we can revive this and get the Ender 2 Pro added? I'm trying to move off Cura and do everything in Orca, but missing a config set for my little printer.

Waynegodli, 2024-01-16:
As above do believe it would be a good addition and i can just transfer to using one slicer

Bcjc3, 2024-01-27:
I agree. I don't use it much, but as a nice mobile "teaching" 3D printer, I would love to see a premade profile for it.

Sjullep, 2024-02-10:
I have 2 of them and i would like to try Orca

balazsbohonyi, 2024-05-17:
It's quite easy to create a profile in Orca Slicer for Ender-2 Pro.
Orca Slicer, initially branched from Bambu Studio and ultimately tracing its origins to renowned slicers like Prusa Slicer, Super Slicer, and Slic3r.

...so the plan is to have Prusa Slicer (Super Slicer, or Bambu Studio) installed - which all have a printer profile for Ender-2 Pro. Configure one of those slicers with that profile, and then you can compare the settings of that profile with the one you will set up in Orca Slicer.

In Orca Slicer, as a starting profile try the Ender-3 Pro profile for example, edit it, and than change all the settings comparing it to the profile from Prusa or Super Slicer. I did this exactly this, and everything's fine.