OrcaSlicer/OrcaSlicer #1641: Build height default incorrect for a Voron Trident 350 with 0.4mm nozzle

oernster, 2023-07-30:
https://github.com/SoftFever/OrcaSlicer/blob/main/resources/profiles/Voron/machine/Voron%20Trident%20350%200.4%20nozzle.json#L18C24-L18C27

The above link has an incorrect build height and this caused me to break my Tap and snap off the build rail attachment by accident.  User error I know but it really needs to be fixed.  Furthermore, the expected height for a Trident 350 is 250 but the likely max is 245.

Therefore, please can you set it to 245mm for the build height.

neeamradia, 2023-07-30:
This issue likely exists with the 300 and 200 trident profiles too.

AlbinoDeer, 2023-07-30:
All builds should be 250mm (stock Voron config)

SoftFever, 2023-07-31:
@oernster 
Ouch!
Somehow I was under the impression that Trident is similar to 2.4 but with about 10-20 mm less Z height.
Boy, I was wrong.

Hope your printer is fine now (hey, it's battle proven now ;)

SoftFever, 2023-07-31:
Fixed.

shaneshort, 2023-08-04:
It's worth noting that the LDO spec trident has a much higher Z volume, which may be where you got your information from. I don't know if it's worth having an extra profile for it or not though.

closed 2023-11-21 by github-actions[bot]