OrcaSlicer/OrcaSlicer #1642: Overhangs now only in classic mode? Speed is not uniform

FillPavS, 2023-07-31:
**Describe the bug**
Part of the overhang perimeter now uses the bridge speed and part of the overhang speed. This happened before if you turn on the classic mode. If you do not use the classic mode, then the speed of the perimeters of the overhangs was taken from the "Overhangs speed" for the entire perimeter. It worked fine. Now the choice of the classic mode is gone and it seems to me that it is used by default.

**3mf File for This Bug**
(https://cloud.mail.ru/public/KubT/nNmSkiLFP)

**To Reproduce**
Slice my model and see the speed

**Expected behavior**
The same speed must be applied to the entire perimeter on the overhang otherwise it causes artifacts. This worked before

**Screenshots**
Last version from Actions - https://github.com/SoftFever/OrcaSlicer/actions/runs/5707259516
[image]

Version before - https://github.com/SoftFever/OrcaSlicer/actions/runs/5656154238
without classic mode
[image]

with classic mode
[image]

**Printer model**
Bambu Lab X1C

**Desktop (please complete the following information):**
 - OS: Windows 10
 - Last version from Actions - https://github.com/SoftFever/OrcaSlicer/actions/runs/5707259516

nagisasatoru36, 2023-07-31:
I was creating a profile for my FLSUN Super Racer and speed seemed inconsistant as well. I check the FLSUN V400 profile and the speed behaviour is strange :

[image]

That infill speed is just ridiculous and doesn't match profile speed.

FillPavS, 2023-07-31:
@nagisasatoru36 You have a different problem. In the filament settings, on the cooling tab, uncheck "Slow printing down for better layer cooling". But then the plastic may not have time to cool down and this will affect the quality of the print.

nagisasatoru36, 2023-07-31:
> @nagisasatoru36 You have a different problem. In the filament settings, on the cooling tab, uncheck "Slow printing down for better layer cooling". But then the plastic may not have time to cool down and this will affect the quality of the print.

That woud not slow down the print that much anyway especially those infill at 30mm/s.
But just by removing ironing without touching filament cooling resolve the issue! Very strange behaviour : 

[image]

Noisyfox, 2023-08-02:
> That woud not slow down the print that much anyway especially those infill at 30mm/s. But just by removing ironing without touching filament cooling resolve the issue! Very strange behaviour : [...]

Could you share your project file?

closed 2023-11-21 by github-actions[bot]