OrcaSlicer/OrcaSlicer #1644: Calibration test don't allow values in the wiki itself

nagisasatoru36, 2023-07-31:
Bug : 

During the calibration of retraction (from calibration tab) the default values (also shown in the wiki) isn't allowed (0 mm start / 2mm end / 
0.1mm/mm step)

[image]

And the prompt showing error doesn't match the wiki and default ; Start > 0 is not verified in default and wiki (here  >= 0). Plus there a flaw in the prompt : step can be equal to 0 (meaning no increment during test ?? So no test at all if increment = 0)

And even after verified prompt request it doesn't work : 

[image]

Reproduce : 

Launch a calibration test.

Expected behavior : 
 
That you could manage a retractation calibration test with default values/coherent values.

SoftFever, 2023-07-31:
@nagisasatoru36 
if you are using France language, the 0.1 step number should be `0,1`, not `0.1`
[image]

nagisasatoru36, 2023-07-31:
> @nagisasatoru36 if you are using France language, the 0.1 step number should be `0,1`, not `0.1` [image]

Indeed! It works with colonl!

[image]

But that a poor way to code user input in that scenario. Other input ignore the '.' and replace it with ',' in profile (for example if i set 0.2 at layer height and press [enter] it becomes 0,2 and accept value. So that a langage isssue (We french aren't moron and understand the '.' syntax lol. Maybe we should just set '.' as every single french program that I use anyway).

So here is a langage bug related.

SoftFever, 2023-07-31:
@nagisasatoru36 
Hahaha, that will definitely make the programmer's life easier ;)
I actually released a hotfix to fix this exact issue
[image]

Joke aside, I believe it reads your system's setting, if you change the Decimal symbol to `.` in your system setting, it should be automatically accept `.`. instead of `,`.
But I didn't test it myself.
[image]

closed 2023-08-06 by SoftFever