OrcaSlicer/OrcaSlicer #1648: Allow import of Prusa slicer .ini files for filament profiles

MAVProxyUser, 2023-07-31:
It feels remiss that you didn't include a conversion tool to import the plethora of slicer profiles available in the community for Prusa, and make them available to Orca. 

Comparing the two we can see similarities. 
https://raw.githubusercontent.com/SoftFever/OrcaSlicer/6c8b01199107dae1bbb0a10cbf337b1795e5faea/src/libslic3r/Print.cpp

https://raw.githubusercontent.com/prusa3d/PrusaSlicer/287e1e28cd6de02585671d7f74d4e36fddd46674/src/libslic3r/Print.cpp

Sure, let's specifically compare the `steps_gcode` sets in both snippets to identify the differences related to the "BBS" printer model (Bambu Studio) in the code:

Here are the common `steps_gcode` entries in both snippets:

- `"autoemit_temperature_commands"`
- `"avoid_crossing_perimeters"`
- `"avoid_crossing_perimeters_max_detour"`
- `"bed_shape"`
- `"bed_temperature"`
- `"before_layer_gcode"`
- `"between_objects_gcode"`
- `"bridge_acceleration"`
- `"bridge_fan_speed"`
- `"enable_dynamic_fan_speeds"`
- `"overhang_fan_speed_0"`
- `"overhang_fan_speed_1"`
- `"overhang_fan_speed_2"`
- `"overhang_fan_speed_3"`
- `"colorprint_heights"`
- `"cooling"`
- `"default_acceleration"`
- `"deretract_speed"`
- `"disable_fan_first_layers"`
- `"duplicate_distance"`
- `"end_gcode"`
- `"end_filament_gcode"`
- `"external_perimeter_acceleration"`
- `"extrusion_axis"`
- `"extruder_clearance_height"`
- `"extruder_clearance_radius"`
- `"extruder_colour"`
- `"extruder_offset"`
- `"extrusion_multiplier"`
- `"fan_always_on"`
- `"fan_below_layer_time"`
- `"full_fan_speed_layer"`
- `"filament_colour"`
- `"filament_diameter"`
- `"filament_density"`
- `"filament_notes"`
- `"filament_cost"`
- `"filament_spool_weight"`
- `"first_layer_acceleration"`
- `"first_layer_acceleration_over_raft"`
- `"first_layer_bed_temperature"`
- `"first_layer_speed_over_raft"`
- `"gcode_comments"`
- `"gcode_label_objects"`
- `"infill_acceleration"`
- `"layer_gcode"`
- `"min_fan_speed"`
- `"max_fan_speed"`
- `"max_print_height"`
- `"min_print_speed"`
- `"max_print_speed"`
- `"max_volumetric_speed"`
- `"notes"`
- `"only_retract_when_crossing_perimeters"`
- `"output_filename_format"`
- `"perimeter_acceleration"`
- `"post_process"`
- `"gcode_substitutions"`
- `"printer_notes"`
- `"retract_before_travel"`
- `"retract_before_wipe"`
- `"retract_layer_change"`
- `"retract_length"`
- `"retract_length_toolchange"`
- `"retract_lift"`
- `"retract_lift_above"`
- `"retract_lift_below"`
- `"retract_restart_extra"`
- `"retract_restart_extra_toolchange"`
- `"retract_speed"`
- `"single_extruder_multi_material_priming"`
- `"slowdown_below_layer_time"`
- `"solid_infill_acceleration"`
- `"standby_temperature_delta"`
- `"start_gcode"`
- `"start_filament_gcode"`
- `"toolchange_gcode"`
- `"top_solid_infill_acceleration"`
- `"travel_acceleration"`
- `"thumbnails"`
- `"thumbnails_format"`
- `"use_firmware_retraction"`
- `"use_relative_e_distances"`
- `"use_volumetric_e"`
- `"variable_layer_height"`
- `"wipe"`

Entries specific to the first snippet (not found in the second snippet):

- `"filament_soluble"`

Entries specific to the second snippet (not found in the first snippet):

- `"additional_cooling_fan_speed"`
- `"reduce_crossing_wall"`
- `"max_travel_detour_distance"`
- `"printable_area"`
- `"bed_exclude_area"`
- `"before_layer_change_gcode"`
- `"enable_pressure_advance"`
- `"pressure_advance"`
- `"enable_overhang_bridge_fan"`
- `"overhang_fan_speed"`
- `"overhang_fan_threshold"`
- `"slow_down_for_layer_cooling"`
- `"extruder_clearance_height_to_rod"`
- `"extruder_clearance_height_to_lid"`
- `"filament_flow_ratio"`
- `"reduce_fan_stop_start_freq"`
- `"fan_cooling_layer_time"`
- `"full_fan_speed_layer"`
- `"fan_kickstart"`
- `"fan_speedup_overhangs"`
- `"fan_speedup_time"`
- `"default_filament_colour"`
- `"outer_wall_acceleration"`
- `"inner_wall_acceleration"`
- `"initial_layer_acceleration"`
- `"top_surface_acceleration"`
- `"bridge_acceleration"`
- `"travel_acceleration"`
- `"sparse_infill_acceleration"`
- `"internal_solid_infill_acceleration"`
- `"cool_plate_temp_initial_layer"`
- `"eng_plate_temp_initial_layer"`
- `"hot_plate_temp_initial_layer"`
- `"textured_plate_temp_initial_layer"`
- `"gcode_add_line_number"`
- `"layer_change_gcode"`
- `"fan_min_speed"`
- `"fan_max_speed"`
- `"printable_height"`
- `"slow_down_min_speed"`
- `"reduce_infill_retraction"`
- `"filename_format"`
- `"retraction_minimum_travel"`
- `"retract_when_changing_layer"`
- `"retraction_speed"`
- `"use_firmware_retraction"`
- `"slow_down_layer_time"`
- `"standby_temperature_delta"`
- `"machine_start_gcode"`
- `"filament_start_gcode"`
- `"change_filament_gcode"`
- `"wipe"`
- `"wipe_distance"`
- `"curr_bed_type"`
- `"nozzle_volume"`
- `"chamber_temperature"`
- `"thumbnails"`
- `"nozzle_hrc"`
- `"required_nozzle_HRC"`
- `"upward_compatible_machine"`
- `"seam_gap"`
- `"role_based_wipe_speed"`
- `"wipe_speed"`
- `"use_relative_e_distances"`
- `"accel_to_decel_enable"`
- `"accel_to_decel_factor"`
- `"wipe_on_loops"`
- `"exclude_object"`
- `"support_material_interface_fan_speed"`

As we can see, the second snippet (Bambu Studio, or "BBS") includes many additional configuration options specific to their printer model or features, which are not present in the first snippet. These additional options are mainly related to cooling, bed and plate temperatures, extruder clearance, filament properties, gcode control, nozzle behavior, and other printer-specific settings.

On the other hand, the first snippet does include one option, `"filament_soluble"`, which is not found in the second snippet. This option seems to be related to support material and soluble filament.

These differences in the `steps_gcode` sets reflect the variations in supported options and printer-specific features between the two versions of the code. Each version caters to the specific needs and functionalities of its respective printer model (OrcaSlicer and Bambu Studio).

How can we get a tool to convert the supported elements between both from .ini to .json?

theophile, 2023-08-03:
https://github.com/SoftFever/OrcaSlicer/discussions/1636#discussion-5455618

Just about done with a version that does print profiles too.

MAVProxyUser, 2023-08-03:
@theophile your base work is great for the Filaments... let me know when you get the print profile too, I've got one I need to convert! Are you also going to do printer profiles, or just the print?

MAVProxyUser, 2023-11-06:
make this fresh...

ALEX-9-99, 2023-12-03:
It would be great if they had some kind of converter layer. I need the whole thing for PP printing with my own profiles.

SPAZv, 2023-12-05:
Im waiting too, maybe will be posibility yo move prusa profiles to orca :)

theophile, 2023-12-05:
In the meantime, see if this works for you:
https://github.com/theophile/SuperSlicer_to_Orca_scripts/

SPAZv, 2023-12-05:
Im not that good in programing language and i try it before but dont know
how to do it right :)

closed 2024-03-12 by github-actions[bot]