OrcaSlicer/OrcaSlicer #1649: [FR] Allow wiping alongside a high "Travel Distance Threshold"

killerline10168, 2023-07-31:
**Is your feature request related to a problem? Please describe.**
I find Cura's style of wiping to be the optimal one, which is the only thing holding me back from changing. Especially since the effect on print quality is huge due to the wipes. In Orca, wiping works only when the travel distance is longer than "Travel Distance Threshold" (a.k.a. THD), but then the slicer forces a retraction as well, which gives weird effects on my prints and ruins them sometimes...

**Describe the solution you'd like**
My suggestion to solve this: imagine printing a cylinder with multiple walls. We don't want the printer to retract just for jumping to the next wall, but if TDH is increased it won't retract, but it also won't wipe anything, which is what I want and am suggesting. 

Here it doesn't retract, but it also doesn't wipe...

[image]

Here it actually wipes, but it also retracts, which is what I don't want.

[image]

Clarification: I wish to be able to increase TDH so the printer doesn't retract for small travel distances, but still be able to make it wipe. I basically want it to always wipe!

Please let me know if I am not clear enough! :)

killerline10168, 2023-08-12:
Nothing? At least have the courtesy to respond to me if anything is wrong or if there is a lack of time.

killerline10168, 2023-11-12:
@SoftFever would adding an "Always wipe" option be difficult?

SoftFever, 2023-11-12:
Hi @killerline10168 
Do you have some photo of the print for comparison to showcase the quality difference?

killerline10168, 2023-11-13:
> Hi @killerline10168 Do you have some photo of the print for comparison to showcase the quality difference?

Not at the moment. 

Cura does this well by default, but that wouldn't be a great comparison, would it? 
I will try to force Orca to wipe on all walls and try to showcase the difference. However, I would argue that wiping would be more or less helpful depending on the printer and filament type as well as how fine-tuned the printer is. 
 
Let me get back to you with pics when I have spare time this week!

killerline10168, 2023-11-15:
I got some pics to showcase the differences, it should be easy enough to follow along:

We always want as constant flow as possible while printing, right? Imagine we are printing a single object and all we're printing are walls and infill within those walls (imagine a cube-like print). In reality, why would we need to retract the filament and obstruct the nice flow, when it's not something we would see? Here are my quick results comparing three different methods to seams:

[image]

First is no wiping, (which is off by default) but always to retract, even from wall to wall. It looks good IMO, but we can do better. 

[image]

Here both retractions and wipes are enabled everywhere (walls, infill, etc). This looks a bit better and is preferable to the one above, but as mentioned, we really don't need to retract when going up, switching walls, or traveling to infill.

[image]

Here is the final and IMHO best alternative. It may be hard to tell by the picture and the harsh lightning, but this print has the best-looking seam of all the prints, it protrudes the least from the walls (dimensional accuracy - think print in place objects) and it ensures that the printed object will be watertight because the hot nozzle is fuzzing the two ends together more than before. 

From my past experience with Cura and wiping, I would argue that an option to ensure wiping would greatly benefit OrcaSlicer. Cura's "Outer Wall Wipe Distance" will wipe all outer/visible walls, which is great for surface quality, but wiping inner walls as well will ensure a stronger bond between wall ends and enhance object water tightness. The same arguments apply to other line types.

As wipes are already present in OrcaSlicer, I imagine that this should not take a lot of work to implement. Opinions?

killerline10168, 2023-11-22:
Has a dev had the time to look at this yet?

killerline10168, 2023-12-07:
@SoftFever Bumping so the ticket won't close as inactive.

igiannakas, 2023-12-13:
So what exactly are you looking for in this? To clarify:
1. Wipe always on external walls
2. Retract at the end of the wipe if there is any filament left to retract

or wipe on the external walls with the filament stationary in the extruder? (ie no retraction while wiping and just let the filament ooze out while doing the wipe move?)

killerline10168, 2023-12-13:
> So what exactly are you looking for in this? To clarify:
>
> 1. Wipe always on external walls
> 2. Retract at the end of the wipe if there is any filament left to retract
>
> or wipe on the external walls [...]

There are a lot of parameters to decide if a retraction should be made. For optimal seam quality, external (i.e. visible) walls should always be wiped with the filament stationary in the extruder (to relieve the pressure). IMO it should only retract afterwards if necessary, like travel moves longer than "Travel distance threshold", or depending on the "Reduce infill retraction" setting. 

I am not sure if we need to wipe internal walls as well, what do you think? From my observations, wipes help more than they hinder.

igiannakas, 2023-12-13:
let me have a think about it and see how this can be done/if it can be done in a straightforward manner. 

In its core Orca slicer has many components from Prusa slicer and Slic3r which always link the wipe move to a retract move. This may need some significant refactoring of the code if we want to always perform a wipe move irrespective of if a retraction is needed. I'll take a look at this in more detail next week when more free from work.

killerline10168, 2023-12-13:
> let me have a think about it and see how this can be done/if it can be done in a straightforward manner.
>
> In its core Orca slicer has many components from Prusa slicer and Slic3r which always link [...]

I understand. Thank you for taking the time to look into this and let me know if I can contribute in any way or perform more tests! 

Just for fun, what is your view on the matter? Do you agree with my results? Do you have other experiences with this?

igiannakas, 2023-12-13:
Well I haven't noticed much of an impact of the wipe operation on my prints - however I am running a Bambu and a klipperised ender 3 s1 which both can control filament flow really well with Pressure advance tuned, so there is very little effect with these options as there is little/no pressure left in the nozzle when its told to stop extruding.

I can see the benefit of retracting while wiping and finishing it off with a retract move if needed on all external walls. It would allow for any spill over filament to be wiped and also smooth out the seam a little bit. But I would think that wiping with the extruder stationary would cause issues especially with printers without PA as there would be significant flow happening after the extruder stopped extruding if we did not retract the filament at all while wiping.

killerline10168, 2023-12-13:
> Well I haven't noticed much of an impact of the wipe operation on my prints - however I am running a Bambu and a klipperised ender 3 s1 which both can control filament flow really well with Pressure [...]

Understandably, this feature would work better with printers running Bowden, such as mine. I am not sure if Bowden will die out, but if this feature requires too much work, then perhaps dropping it would be better...

> I can see the benefit of retracting while wiping and finishing it off with a retract move if needed on all external walls. It would allow for any spill over filament to be wiped and also smooth out [...]

Would it cause issues? That would surprise me, as that is the way Cura does it by default (if I am not mistaken) and what I am trying to bring over to Orca xD:

[image]

https://github.com/SoftFever/OrcaSlicer/assets/1812366/3e53154e-90ca-4435-ad6c-9c06326e946e

Blue lines represent travel moves without retraction, red is the outer wall and green is the inner wall.

igiannakas, 2023-12-13:
Take a look at this:

https://www.reddit.com/r/Cura/s/ja6lPkHdZZ

https://drive.google.com/file/d/1MAk4xWxDtPS_uSHvGQRyr2Z6LSfXMG7H/view

here they don’t want the cura method 🫣

killerline10168, 2023-12-13:
Just downloaded your PR from the other ticket. The way you have set it up seems very reasonable, will give it a try and get back to you.

[image]

igiannakas, 2023-12-14:
That may be a bug - I’ll test it out again

igiannakas, 2023-12-14:
Can you post your project file please?

igiannakas, 2023-12-14:
Can't replicate with the latest build:

[image]

https://github.com/SoftFever/OrcaSlicer/actions/runs/7198425797

killerline10168, 2023-12-14:
> Can't replicate with the latest build: [...]

Saw that some changes has been made and that a new PR is being built as I type. I will try it out once finished and let you know the results.

FlowForm, 2024-02-18:
I have the same issue here, with several filaments I get a much better seam when there is a wipe. 
I cannot get this at all layers, unless I use the "Aligned" in seam position, but this makes the seam wobly compared to "Back" as you can see in the screenshots. 

I would be awesome with an "always wipe on exterior wall" function.!! :-)

[image]

closed 2024-05-27 by github-actions[bot]