OrcaSlicer/OrcaSlicer #1655: Profile Dependencies (similar to what both PrusaSlicer and SuperSlicer have already)

jeremytodd1, 2023-08-01:
Printing profile lists are always getting bigger and bigger, especially when you have multiple printers. 

Dependencies is a very useful feature.
What this does is that it lets you tie certain profiles to specific printers and nozzle sizes, so that way only those specific profiles show when you have that specific printer selected.
Both SuperSlicer and PrusaSlicer have had this for years.

For example, I currently have two Voron 2.4 350mms. One orange, one blue.
If I currently have the orange printer selected, I'm still forced to see the profiles I use for the blue printer. 
These profiles add up quickly so it starts to be quite a list.
With dependencies it's possible to make it so only filament/print profiles I use with the orange printer would show.

I believe this would be a very useful feature to port over to Orca.

Carib3D, 2023-08-01:
I want to support this request for Dependencies. In fact I had made request  at about the same time.

#1654
In SS and PrusaSlicer dependencies can be set so that only specified filaments and processes show up for a given printer. 
As another suggestion, and this might be crazy talk, could another heading be added for Nozzles? So instead of (Printer-Filament-Process) headers, It becomes (Printer-Nozzle-Filament-Process). The Nozzle heading would take the Extruder section of the Printer and parts of the Quality & Support sections of the Process header that are relevant to nozzle size. If this could be implemented with dependencies, I believe it would simplify the workflow for those of us that use multiple printers and multiple nozzles.

yourbuddydinec, 2023-08-24:
I would also like to add my desire for this. I love orcaslicer since I started to use it before but when you have a ton of filaments among multiple printers the filament selection drop down can get a little crazy. So dependencies would be awesome!

jeremytodd1, 2023-11-24:
Commenting to prevent automatic closure

yourbuddydinec, 2023-11-24:
Yes please. Would be very useful to me as well

quantumfrost, 2024-01-11:
Commenting to prevent automatic closure

murdock62, 2024-01-14:
Hello !

Could you please look into this feature? For me, this is the main blocking point when using this very good slicer. Many thanks for your work.

Sixpack73, 2024-01-21:
I would love to see this feature added.   Allow profiles to be dependent on what printer and nozzle size you have selected!

SwineFlew931, 2024-02-18:
I am here to support this request as well. Having multiple printers with vastly different filament profile requirements and 100 plus filaments on hand, my filament profile list is out of control. Dependencies are the only thing I miss about PrusaSlicer. It made it so much cleaner/easier to set up each print. Please?

Aron3DP, 2024-03-26:
This is the only feature from PS/SS that I miss.

Zerufuru, 2024-03-26:
Dependencies would be a huge improvement and absolutely necessary to manage profiles across multiple printers

yourbuddydinec, 2024-03-27:
Yeah I would very much appreciate it. It's been a few times (though rarely) i select a filament with flows/fan speeds for the wrong printer on accident 😩

ramosglauco, 2024-04-26:
Please @SoftFever! Would be a huge improvement 🙏🏻🙏🏻🙏🏻

quantumfrost, 2024-04-26:
We also need filament profile dependencies so the same filament can be used on multiple machines

stavrosg, 2024-05-07:
Woudl love this feature!!!

yellowicz, 2024-06-05:
I like Orca and would love to have specific profiles depending on the choice of printer, nozzle or filament. Due to the lack of this function, I am waiting for the new version of Super Slicer.

lanpartyceo, 2024-06-15:
+1 from me for this Feature Request as well

This alone is the primary reason why I still use PrusaSlicer. I have no choice because orcaslicer does not support dependencies. 

Why is this important? Right now if I want to print on two different making model printers, even if those printers use the same gear ratio on the extruder and same size nozzle and even the same type nozzle, I still have to make a new filament profile specifically for each printer even when the filament itself hasn't changed at all. Well undoubtedly there are certainly good reasons for having filaments set to specific model printers only, More often than not people are using filament from filament production companies which falls outside of that scope. 

Since the film is designed to be used on any printer and the filament settings in terms of the temperature of melts at and the flow rate and the bed temperatures are all going to be the same regardless of what printer I'm printing on so having to rebuild the profile each time just because I switched from one printer to another is absolutely ridiculous the majority of the time. 

Allowing the creation of a profile where I can then edit the dependencies of that filament that let it work with any printer would save a vast amount of time and headache.

ThinkBigPrintHuge, 2024-11-01:
+1
I would love to see filament and processes coupled.

ramosglauco, 2024-11-04:
+1
Please, why hasn't this been implemented yet? Should be priority number 1

Greecher, 2024-12-11:
+1
I've been a devout user of dependencies to keep screw ups to a minimum, please consider this, as this is the only item I really really need to keep sanity.

CMarkussen, 2024-12-23:
+1 on this feature. It is a pain to maintain profiles for multiple nozzle sizes and dependencies would help this with parameters. Alternatively, splitting out the nozzle size parameters as a separate view would be really useful.

closed 2025-01-13 by SoftFever

ThinkBigPrintHuge, 2025-03-20:
> Maybe i don't get the big picture, but why are the dependencies tied only to the filaments? E.g. i want to set different accelerations for different printers. This has nothing to do with filaments but with the current implementation i have to duplicate a lot of filament profiles just for this purpose.

Agreed. I get so lost trying to build profiles in Orca. I wish it was handled the same way S3D handles profile building.

SoftFever, 2025-03-29:
> Maybe i don't get the big picture, but why are the dependencies tied only to the filaments? E.g. i want to set different accelerations for different printers. This has nothing to do with filaments [...]

It can support the process as well. It's just that there are some UI glitches and I can't find a better way to make it work nicely. So I disabled it. And also I think processes compared to filament are more machine-correlated. Hence, it's not that common to share processes across machines.

Anthrax-VX, 2025-05-12:
This should be a very simple setup, make it all dependent on the printer you have selected.  in each process and filament, you choose their compatible things, like filaments you choose compatible printer, then compatible processes.  for processes, you choose compatible printers, and compatible filaments, shouldn't be that hard, that way when you select the printer, it will only show the filament settings that are associated with that printer, then when you select the filament, it only shows the processes that work for that printer and that filament.

PR #6235 (2025-01-13): enabled dependency setting for filament and process
# Description
Enabled dependency setting for filament and process for **Developer mode**. Dependency feature seems to be commented out by BBS, this PR is just to uncommenting dependency tab related code. This fixes #1655  *This is probably not release-ready, but I want to share what I have.

Only changes I added were
from `toggle_options();`
to
`    if (m_active_page && !(m_active_page->title() == "Dependencies"))
        toggle_options();`
This is necessary to prevent app crashing from viewing Dependency tab.

edit: also fixes #3497 

# Scre [... cut]