OrcaSlicer/OrcaSlicer #1660: Feature request - sync PA value from filament to the printer profiles?

zviratko, 2023-08-02:
**Is your feature request related to a problem? Please describe.**
Latest Bambu Studio and firmware has the option to calibrate the filament K-Value
But it doesn't seem to save to the .json that is synced but rather pushed to the printer via mqtt (interestingly, I can see the pressure advance value from Orca in the .json in their filament profiles ~/Library/Application Support/BambuStudio/user/xxxx/filament/ which I believe was filtered out before when I looked).

It would be great to be able to keep those profiles and filaments in-sync between OrcaSlicer and Bambu Studio, possibly switch to their way of doing things to avoid problems in the future.  It's probably the way Bambu wants this to be done instead of using G-Code for PA. I don't think they are going to change the way they do it, so it would be good to just use their approach... (I also think it's a better way of doing things)

**Describe the solution you'd like**
I'd like to be able to push the K-Value from the filament profile in OrcaSlicer to the printer with the name of the filament.
Maybe add a button in the filament profile which would push that profile (both Flow Rate Multiplier and K-Value) to the printer like Bambu Studio does after calibration.

Not sure what other changes need to be made to be closer to upstream.

Alternative: Allow us to just save the calibration manually without actually doing it, just taking the last step from their Manual Calibration and using that.

**Describe alternatives you've considered**
We could continue using the G-Code way as with other printers, but I can imagine Bambu is just going to remove PA setting from the G-Code at some point and then we're stuck, or we might lose some functionality.
Sadly, there's no way in Bambu Studio to just "push" a new profile without doing the calibration first, even if you know what value you want in there.
We could ask Bambu to just use the standard way, but I think there might be a good reason behind that - this way you can slice an object using one PLA brand and then reprinting it using a different filament with different K-Value (and possibly Flow Rate Multiplier?) and the printer will just adjust that for you. I wonder if their aim is to be able to switch materials as well. I think we should follow and not get stuck in the past way of doing things...

kylek29, 2023-08-04:
>I can see the pressure advance value from Orca in the .json in their filament profiles ~/Library/Application Support/BambuStudio/user/xxxx/filament/ which I believe was filtered out before when I looked).

Are you sure you never set that value? My filament .json's in Orca have this value because I explicitly set it. If you leave the defaults, they inherit from the upstream profile it was forked from.

zviratko, 2023-08-05:
The point is that the way OrcaSlicer does it and the way Bambu Studio does it are completely different
OrcaSlicer sets the K using gcode commands, Bambu Studio pushes that to the printer and set it for the filament profile.

What this means is that you can grab any sliced model, select material for it, and the printer would apply PA compensation to it, even if it wasn't sliced this way. I think this is great, but it's not what everybody else is doing.
For this to work, OrcaSlicer needs to use the new API.
I have no idea what actually happens on the new firmwares if you use the gcode way of doing things (I assume it will still work), but the new way is just better, and is cross-compatible with Bambu Studio.
Bambu Studio doesn't use PA values from the .json file, at least I don't think so.

kylek29, 2023-08-05:
@zviratko I was just adding additional context to your request and pointing out that in the meantime we can run BambuStudio to get the values generated from the Lidar (not something we could do previously) and then populate those into our Filament profiles.

I haven't looked at the GCode generated, but I would assume BambuStudio drops a hint comment on any values that Flow Rate / Dynamics would affect so that it can do an on the fly replacement when the firmware processes the print. If you open the file being sent to the printer, it is still just standard GCode. This is how they do most of the other specialty things (function comments).

zviratko, 2023-08-05:
I just looked through gcode generated by the latest Bambu Studio and while M900 K0 command is used, it is only used for the extrusion calibration procedure, not during the actual print.

I'm not sure if previous versions used M900 K0 command during the print, but I don't think so as Bambu Studio never had an option to use the calibrated value. AFAIK it was always printer-side feature tied to the filament type.
(But I haven't used BS for some time before the 1.7 version, so not sure how calibration worked before this version).

Bambu also claim that the value obtained from manual calibration and automatic calibration might differ (but somehow both be correct) because of "different algorithm" - I'm not sure if they have another version of PA with a diferent value of K (or rather "not-really-K") or are bluffing when lidar messes up :)

But this is a bit OT.

The purpose of my Feature Request was to be able to leverage what BS does by offloading it to the printer while keeping all the cool features of OrcaSlicer, that means either porting the Calibration portion, or making something else (and better), allowing direct manipulation of the calibrations, or converting the pressure advance from .json into filament profile on the printer or something like that...

SoftFever, 2023-08-06:
Setting PA per filament is the most flexible and sensible way to manage PA values IMHO
I don't see any sense to set PA to slots like what they did.

zviratko, 2023-08-06:
Well, they are setting it for filament, but so far they assume you are doing automatic calibration at print start. When you change filament in that slot it detects it and the PA calibration is reset.
It's a bit clunky, as it doesn't remember that "Bambu PLA-CF has a K=0.02", but I fully expect this to either be bug or just a missing feature coming soon.

It makes sense to me because you can just change filament after slicing and get the correct calibration for it independent on what it was sliced with, in most cases you could even change the material from for example PLA to PA and still get it to print (I think the evidence that this is coming is in there, otherwise there would be no use for the temperature setting for the material on the printer).

So what are your intentions, are you not going to not implement this management at all? You could make it even better than what Studio does, i.e. setting the K automatically for the slot/material upon insertion, allow manual profile creation etc...

SoftFever, 2023-08-07:
I don't have a X1 printer with me.
It's a bit difficult for me to test and work on that part.

No plan to work on this.

closed 2023-08-07 by SoftFever