OrcaSlicer/OrcaSlicer #1663: OrcaSlicer submits telemetry to bambulab.com

matt-e, 2023-08-03:
**Describe the bug**
When I start  OrcaSlicer from the command-line, I see a bunch of HTTP requests and responses logged that go to `e.bambulab.com`. This appears to include a uniquely identifying ID - and my IP address. This feels like the kind of telemetry that should stay out of an open-source project unaffiliated with Bambulab.  It appers the unique ID is generated once and preserved forever across updates, allowing lifetime tracking of my usage. 

There doesn't appear to be an opt-out mechanism nor a privacy policy describing what's done with the data.

For what it's worth, it seems pretty straightforward to disable the telemetry - I assume it's not here on purpose. I could try put up a PR if the consensus is that it should be removed.

Full request below:
```
*   Trying 18.172.170.35:443...
* Connected to e.bambulab.com (18.172.170.35) port 443 (#0)
* ALPN, offering http/1.1
* SSL connection using TLSv1.3 / TLS_AES_128_GCM_SHA256
* ALPN, server accepted to use http/1.1
* Server certificate:
*  subject: CN=*.bambulab.com
*  start date: Jun 15 00:00:00 2023 GMT
*  expire date: Jul 14 23:59:59 2024 GMT
*  issuer: C=US; O=DigiCert, Inc.; CN=RapidSSL Global TLS RSA4096 SHA256 2022 CA1
*  SSL certificate verify result: unable to get local issuer certificate (20), continuing anyway.
> GET /query.php?lang=en HTTP/1.1
Host: e.bambulab.com
User-Agent: OrcaSlicer/01.06.00.66
Accept: */*
X-BBL-Client-Name: OrcaSlicer
X-BBL-Client-Type: slicer
X-BBL-Client-Version: 01.06.00.66
X-BBL-Device-ID: 370f9f43-c6fe-47d7-aec9-5fe5ef7e7673
X-BBL-OS-Type: linux
X-BBL-OS-Version: 6.4.3
```

**To Reproduce**
Start the latest AppImage from the command-line. 

https://github.com/SoftFever/OrcaSlicer/releases/download/v1.6.3/OrcaSlicer_V1.6.3_Linux.zip

**Expected behavior**
No telemetry should be sent to BambuLab.

**Desktop (please complete the following information):**
 - Arch linux/Kernel 6.4.3/KDE/Wayland

MAVProxyUser, 2023-08-03:
See also: https://github.com/SoftFever/OrcaSlicer/blob/main/src/slic3r/GUI/HMS.cpp#L48

MAVProxyUser, 2023-08-03:
also see: 
[image]
https://github.com/SoftFever/OrcaSlicer/blob/main/src/libslic3r/AppConfig.cpp#L79

MAVProxyUser, 2023-08-03:
Seems to be part of the Health Management System. 
https://wiki.bambulab.com/en/x1/troubleshooting/intro-hms

SoftFever, 2023-08-03:
Thanks @MAVProxyUser for digging into this.
It looks like it is used for the slicer to receive printer notifications.

I've turned off privacy tracking several releases ago.
I'll investigate if it makes sense to add an option in Preferences to allow users to turn it off.
Meanwhile, if you notice it's sending other telemetries, do let me know. @matt-e @MAVProxyUser

MAVProxyUser, 2023-08-03:
@SoftFever I didn't put too much effort into it, but I was able to setup a fake https server, and use DNS redirection and confirmed that the code doesn't check SSL validity. I was able to use it to serve a custom ~/Library/Application\ Support/OrcaSlicer/hms/hms_en.json to the app. I didn't look for the other http downloads like for updates, and profile downloads, but we probably want to do SSL enforcement. It seemed like I could also just serve up an update if I wanted to when it checks GitHub. I was looking for opportunities in which this telemetry could be used for potential code execution, and wound up realizing none of the SSL certs are checked.

MAVProxyUser, 2023-08-03:
FWIW... this is the code I modified to test with, oddly enough an old DJI security vulnerability. BambuLabs is of course made up of old DJI employees, so there is some irony in my reuse of this old code to test. https://github.com/MAVProxyUser/DJI-Unobtanium/blob/master/Unobtanium.rb

R4toLoko, 2023-08-04:
I did noticed this when investigating a javascript error with the slicer latest release after a Linux update.
I didn't take that too seriously. I actually thought it was a normal process checking GitHub for update.

[image]

In my case, it failed because I run Pi-Hole and Firewall rules on OPNSense to block data collection everywhere within my network. I guess that is why I ignored it.

[image]

I also blocked `api.bambulab.com` because I don't have its printer anyway and the slicer works just fine.
And because all the endpoints were blacklisted, it falls back to GitHub itself.

[image]

[image]

closed 2023-08-10 by SoftFever

thijstriemstra, 2023-08-18:
> this is kind of a false positive.

how? there was telemetry being sent right?

> closed as not planned

you stopped telemetey feom working with your commit so this ticket was actually fixed.

Oh well.

SoftFever, 2023-08-20:
> > this is kind of a false positive.
> 
> how? there was telemetry being sent right?
> 
> > closed as not planned
> 
> you stopped telemetey feom working with your commit so this ticket was actually fixed.
> 
> Oh well.

By "false positive", I mean that the e.bambulab.com issue mentioned in the original post is related to the HMS service (which receives updates from printers). Technically, it's not telemetry.
Orca turned off telemetry a few releases back.
In the recent commit, I merely added more locks to it. But technically there should be no difference as the existing version shouldn't send telemetry anyway.
I hope that clarifies things.