OrcaSlicer/OrcaSlicer #1666: Mintion Beagle Cam

Levin08, 2023-08-03:
It would be great if the Beagle Cam could be integrated as well, as it would allow me to use all printers with a single slicer.

I'm not sure what the issue is, as I'm not experienced in programming or figuring out gcode. From what I see on the camera, it doesn't seem to detect the layer height. 
[image]

I don't know if this is a bug or just not supported, as it seems to work with other slicers.

Other than that, I would be delighted if the Beagle Cam's problem or settings could be integrated, or if there is already a solution, I would love to know about it.

SoftFever, 2023-08-04:
@Levin08 
Looks like Beagle Cam is not an open source project?
It would require them to add support to OrcaSlicer.

Levin08, 2023-08-10:
Thank you for your effort, but I've already written an email myself and I am testing the beta. Thank you very much.

SwineFlew931, 2023-10-13:
> Thank you for your effort, but I've already written an email myself and I am testing the beta. Thank you very much.

How is this working out for you? I just Downloaded Orca and am working on profiles. I contacted Mintion, got the Beta and have it on both of my Beagles. Currently printing temp tower calibrations. It seems to be working fine. Any issues for you?

closed 2024-01-19 by github-actions[bot]