OrcaSlicer/OrcaSlicer #1668: Import over new BambuStudio features

npoulos, 2023-08-03:
Would love the new features released by Bambu to be in Orca Slicer.

The stopping printing of certain objects is perfect! Are you planning on merging it over?

tripp396, 2023-08-03:
That's the main feature I'd love to see, especially if it can get past the limitation of multi color printing removed.

igiannakas, 2023-08-03:
I’d love to see the below ported over:
1. Object exclusion
2. Selection of line types for solid infill
3. Automated calibrations for PA and Flow rate

TexZeTech, 2023-08-03:
>The stopping printing of certain objects is perfect! Are you planning on merging it over? 
https://github.com/SoftFever/OrcaSlicer/issues/1668#issue-1834988362

>1.Object exclusion
https://github.com/SoftFever/OrcaSlicer/issues/1668#issuecomment-1664460759

Object exclusion is in the Orcaslicer already no? 

[image]

>2. Selection of line types for solid infill

Elaborate, please?

>3. Automated calibrations for PA and Flow rate

I'm assuming this is the lidar system on the bambulab X1 and other Bambu printers. 
Unless you want to make a system like https://www.youtube.com/watch?v=c1hrP0gduRU for whatever printer you are using.

I would think the code for that would already be in Orcaslicer but I don't have a Bambu printer so I wouldn't know :shrug:

KAsp3rd, 2023-08-04:
> > Object exclusion is in the Orcaslicer already no?
> 
> I believe @npoulos is referring to the new X1 series firmware option to halt printing a single object on the build plate mid-print.
> 
> https://wiki.bambulab.com/en/x1/manual/X1-X1C-AMS-firmware-release-history#added-support-for-model-objects-being-skipped-when-printing

That's exactly how exclude objects works if it's being used with Klipper

TexZeTech, 2023-08-04:
That was kinda my thought as well, granted I haven't used it a lot yet.

kylek29, 2023-08-04:
With regards to the "Exclude" objects feature (being able to do it on the printer from a previous print).

It seems BambuStudio 1.7+ adds the following to the .Gcode:

@ Header of file, directly after "; total layer number: ##":
`; model label id: 78,110,141,162`

@ Each start of a model instance (note, the ID number is for the model, not the instance of the model, so a copy of the same model would produce the same ID):
`; start printing object, unique label id: 78`

@ Each end of a model instance (same rules apply):
`; stop printing object, unique label id: 78`

zoroseerus, 2023-08-07:
> > The stopping printing of certain objects is perfect! Are you planning on merging it over?
> > [#1668 (comment)](https://github.com/SoftFever/OrcaSlicer/issues/1668#issue-1834988362)
> 
> > 1.Object exclusion
> > [#1668 (comment)](https://github.com/SoftFever/OrcaSlicer/issues/1668#issuecomment-1664460759)
> 
> Object exclusion is in the Orcaslicer already no?
> 
> [image]
> 
> > 2. Selection of line types for solid infill
> 
> Elaborate, please?
> 
> > 3. Automated calibrations for PA and Flow rate
> 
> I'm assuming this is the lidar system on the bambulab X1 and other Bambu printers. Unless you want to make a system like https://www.youtube.com/watch?v=c1hrP0gduRU for whatever printer you are using.
> 
> I would think the code for that would already be in Orcaslicer but I don't have a Bambu printer so I wouldn't know 🤷

for #2 https://github.com/bambulab/BambuStudio/issues/945 this. The ability to choose the Solid Infill Type. It got added to BS Beta 1.7 (I think) recently). I believe this post says it came from SuperSlic3r

igiannakas, 2023-08-10:
I see the latest 

> I’d love to see the below ported over:
> 
> 1. Object exclusion
> 2. Selection of line types for solid infill
> 3. Automated calibrations for PA and Flow rate

I see that number (2) above is now implemented in the latest source code :D I've built it locally as I can't wait for this feature to make its way in the official build

Would be lovely to see (1) for Bambu printers. The current object exclusion isn't compatible with bambu printers.

closed 2023-11-21 by github-actions[bot]