OrcaSlicer/OrcaSlicer #1669: Dynamic overhang speeds in 1.6.4 seem to be not working properly

EklDuderino, 2023-08-03:
**Describe the bug**
In 1.6.4 with dynamic overhang cooling settings enabled at 50%, fan speed is not increased for overhangs like in 1.6.3

**To Reproduce**
Slice file with 50% settings for overhang cooling in 1.6.3 and 1.6.4, same file, same settings.

**Expected behavior**
Fan speeds increase for overhangs similarly in both versions 

**Screenshots**
[image]

**Printer model**
Neptune 3 Plus
**Desktop (please complete the following information):**
 - OS: Windows 10

SoftFever, 2023-08-03:
@EklDuderino 
Thanks for reporting.
Is it convient to share the project?

EklDuderino, 2023-08-03:
It is a paid for file, so I am unable to share. I will attempt to recreate the issue with a project I can share. Thank you.

EklDuderino, 2023-08-03:
> @EklDuderino Thanks for reporting. Is it convient to share the project?

I cannot reproduce the issue. Both versions seem to be enforcing fan speed similarly on overhangs now. Must have been an error on my part. Although I am getting a single red 100% line in 1.6.4 on every print I slice. Probably not a big deal, just strange.
[image]

SoftFever, 2023-08-04:
I noticed that too.
Gonna spend some  time to investigate it tonight.

pennystockss, 2023-08-05:
Same here, sliced a model and no overhang slow down. If I slice it with Classic, it does slow down on overhangs. But sliced with Arachne it does not slow down whatsoever.

SoftFever, 2023-08-06:
@pennystockss when version are you using? can you share the project file?

pennystockss, 2023-08-06:
@SoftFever 1.6.4 beta 2. After some messing around it looks like when you turn supports on, it ignores the slow down. But only on Arachne. So:

Arachne: Fine
Arachne + Support: No slow down
Classic +-Support: Fine

[image]
[image]
[image]

SoftFever, 2023-08-10:
> @SoftFever 1.6.4 beta 2. After some messing around it looks like when you turn supports on, it ignores the slow down. But only on Arachne. So:
> 
> Arachne: Fine Arachne + Support: No slow down Classic +-Support: Fine
> 
> [image] [image] [image]

This is different issue.
Could you log a bug for this?
I will fixed in next release

pennystockss, 2023-08-10:
@SoftFever Opps this one is for fan speed only. Ok will make new issue for print speed slowdowns.

pennystockss, 2023-08-10:
@EklDuderino Is it a color print by chance? I'm noticing weird fan speeds if the print is colored.

EklDuderino, 2023-08-10:
> @EklDuderino Is it a color print by chance? I'm noticing weird fan speeds if the print is colored.

All screenshots I uploaded were for single color prints

closed 2023-11-21 by github-actions[bot]