OrcaSlicer/OrcaSlicer #1670: Any plans to release a Flatpak version of OrcaSlicer?

IgnisFlamme, 2023-08-03:
It would be great because downloading .AppImages from Github whenever I want to update is a pain.

pottedplant, 2023-08-12:
Also the AppImage does not work on all distros or at least requires to manually install a bunch of stuff. `flatpak` for e.g. PrusaSlicer just works for me.

closed 2023-11-21 by github-actions[bot]

m0d3rnX, 2023-12-30:
Reopen, the AppImage isn't running on at least Arch and OpenSUSE.

kinghat, 2024-01-12:
looks like bambu was flatpak'd: https://github.com/flathub/com.bambulab.BambuStudio

JaneSmith, 2024-01-14:
As a Fedora Silverblue user, I'd love a Flatpak version. Flatpak is basically the native package manager on my distro. Currently only Cura and PrusaSlicer are available (and I guess Bambu Studio as of very recently).

hz61p1, 2024-01-15:
I second that flatpak is preferable

xpcone, 2024-01-29:
AppImage ist not working on openSuse, Arch, Gentoo, Fedora

woolmonkey, 2024-03-23:
Yes please Appimages suck they are universal at all.

advlogic, 2024-04-30:
Completely agree.  A Fedora package would be great as well.

kejar31, 2024-05-20:
may be a little late to the request but yes, please release a flatpack rather than appimage

TobiZog, 2024-05-30:
I'd like to see a Flatpak version of Orca Slicer too!

powpingdone, 2024-05-30:
#4175 @TobiZog

Anthony-Bec, 2025-01-03:
for some reason appimages dont work for me in kde neon

chrisf4lc0n, 2025-01-25:
> Reopen, the AppImage isn't running on at least Arch and OpenSUSE.

I have the same issue with AppImage and Arch, try to run it from the console and see what dependencies are missing, I cannot remember what was missing but there were 2, once installed it has started without any issues.

Prazgreen1, 2025-01-26:
This has been implemented in a nightly build. This will likely come in the next stable release with it becoming available on Flathub. At least, that is my guess.

Prazgreen1, 2025-01-26:
just note that until it is in an official Flathub repo, you will have to uninstall and manually reinstall, but at least it is a  less painful Flatpak, because as someone who used to run the Appimage, I can easily say that Appimages SUCK. Also, with the recent drama around them, I would rather stay completely care from that packaging format if I can.

Prazgreen1, 2025-02-17:
I do know Flatpak is currently here, but it might not work as some people, myself included, have had some issues likely related to gstreamer that breaks the print camera viewer in Orca Slicer's flatpak for some systems. I am currently daily-driving Fedora Workstation 41. I'll see if anything changes with GNOME and/or GNOME Xorg