OrcaSlicer/OrcaSlicer #1673: Pressure Advance Calibration PA Pattern is torn from the print bed

LoboThirteen, 2023-08-03:
Pressure Advance Calibration PA Pattern is not actually printable without performing a Z-hop. After the first layer, it is torn off the printing plate. Unfortunately, the Z-Hop settings are not adopted in the PA Calibration.

ppamment, 2023-08-07:
Same for me. 
Works fine if I generate the pattern gcode from Ellis online generator, but have tried every which way to get the orca one working without success

Edzi11, 2023-08-10:
I actually get it to print just fine but my issue is that theres no difference in each change. The previous line method seems to actually work though. I really wish there was a way to just add a frame to connect all the lines though

tastyratz, 2023-08-10:
I had this problem last week and it was driving me mad! I thought it was a bed adhesion issue. I also found that I couldn't easily see much of a difference between lines and assumed it was the abs-cf being wet and naturally smoothing. I've had good luck with Ellis online generator. Fixing z-hop would make complete sense.

I also noticed that generating this is retaining wall generator settings with Arachne. I'm not sure if this needs to be classic mode with detect thin lines or not (like a traditional PA tower).

I didn't realize you can't just make a per object or print exception to zhop settings until trying to do so just now.

ppamment, 2023-08-11:
This actually bent a hotend at the heatbreak where it enters the heatsink just now from scraping over the lines quite hard repeatedly. No big deal, it's an aliexpress aftermarket one, I have spare heatblocks, but for now I just bent it back into alignment and it seems to work absolutely fine :)
It should respect the zhop settings in the printer profile though. 

I've never touched cpp before, but it looks like perhaps in
src/libslic3r/calib.cpp:
```
std::string CalibPressureAdvance::move_to(Vec2d pt, GCodeWriter& writer, std::string comment)
{
    std::stringstream gcode;

    gcode << writer.retract();
    gcode << writer.travel_to_xy(pt, comment);
    gcode << writer.unretract();

    m_last_pos = Vec3d(pt.x(), pt.y(), 0);

    return gcode.str(); 
}
```

may want to include calls to lift and unlift?

std::string CalibPressureAdvance::move_to(Vec2d pt, GCodeWriter& writer, std::string comment)
{
    std::stringstream gcode;

    gcode << writer.left();
    gcode << writer.retract();
    gcode << writer.travel_to_xy(pt, comment);
    gcode << writer.unretract();
    gcode << writer.unlift();

    m_last_pos = Vec3d(pt.x(), pt.y(), 0);

    return gcode.str(); 
}

Lift conditions should probably be checked though as implemented in GCode::retract

Perhaps if I have some time to figure out how C++ works this weekend I can raise a PR

closed 2023-11-21 by github-actions[bot]

TimVosch, 2025-01-07:
Just wanted to chime in and say that this still appears to be an issue. The PA pattern from orca has severe dragging issues, while Ellis' generator has not.