OrcaSlicer/OrcaSlicer #1675: 1.6.4-beta fails to connect to printer with TLS errors: "error:1012606B:elliptic curve routines:EC_POINT_set_affine_coordinates:point is not on curve"

samwiseg0, 2023-08-03:
**Describe the bug**
Connecting to a server using TLS fails to connect and be able to upload a gcode file. There was no issue on version 1.6.3.

There is no issue viewing the web interface from the device tab. 

Full error:
```
Could not connect to OctoPrint: curl:SSL connect error:
error:1012606B:elliptic curve routines:EC_POINT_set_affine_coordinates:point is not on curve
[Error 35]
Note: OctoPrint version at least 1.1.0 is required.
```

**3mf File for This Bug**
N/A

**To Reproduce**
Not entirely sure the specifics as I have checked with other folks that use TLS and they have not experienced the same issue. I have also taken a packet capture and it indicates that the error is after the server key exchange and Orca sends back a fatal error to the webserver.

[image]

**Expected behavior**
OrcaSlicer connects to a printer running TLS

**Screenshots**
v1.6.3
[image]

v1.6.4
[image]

**Printer model**
N/A
**Desktop (please complete the following information):**
 - OS: MacOS 13.4.1 (c) (22F770820d)

SoftFever, 2023-08-04:
Hmm, that's odd. I didn't update OpenSSL.

One recent change that might be relevant is PR #1157. 
Maybe @ammmze can shed some light on this matter?

If you can compile from source, try reverting that change to see if the issue persists. 
It would be very helpful to pin-point the issue as I don't have the eviroment setup to test this.

ammmze, 2023-08-04:
Hmm, I'm not sure how disabling ATS would cause that error. But I'll do some digging when I can get to my computer. 

Another potentially relevant issue:
* https://github.com/openssl/openssl/issues/20937

ammmze, 2023-08-04:
So far I have **not** been able to reproduce this. My cert is issued by Let's Encrypt, so I am not using a custom CA cert.

I am running on an M1 Macbook running MacOS 13.5.

Also for reference:

```
$ clang --version
Apple clang version 14.0.3 (clang-1403.0.22.14.1)
Target: arm64-apple-darwin22.6.0
Thread model: posix
InstalledDir: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin
```

```
$ xcodebuild -version
Xcode 14.3.1
Build version 14E300c
```

ammmze, 2023-08-04:
I also just tested this on another Macbook, running an Intel processor and it worked fine as well. Also on MacOS 13.5.

The clang and xcode versions were the same, but OpenSSL version was different FWIW:

```
$ openssl version
OpenSSL 3.1.1 30 May 2023 (Library: OpenSSL 3.1.1 30 May 2023)
```

samwiseg0, 2023-08-04:
@ammmze @SoftFever I plan on building it from source and testing. 

In the meantime I can privately share with you a public URL that should show you the error. I can PM you via discord. I joined the OrcaSlicer discord. Feel free to PM me as my username is the same. I tried looking for your usernames but they maybe different.

ammmze, 2023-08-04:
I will try to do some more debugging tonight, but just to keep everybody updated, I have been able to re-produce the error when using the url provided via PM. So that rules out the possibility that it is isolated to the one machine running OrcaSlicer. 

Another thing I noticed is that despite receiving the error, the webview in the device tab still renders the page just fine. So the error isn't blocking the web view. So that seems even more unlikely that the issue was caused by disabling ATS as I think ATS really only applies to the web view in OrcaSlicer. I imagine that the error we are seeing would be a problem for everything else outside the webview though, so obviously ignoring the error is NOT an option.

SoftFever, 2023-08-05:
Thank you both!

By the way, I'd like to clarify that OrcaSlicer currently doesn't have an official social platform. 
However, I'm in the process of setting up a Discord server.
Will keep you posted!

samwiseg0, 2023-08-05:
I have good news and bad news.

Good news is after building dependencies and OrcaSlicer from source the issue is fixed. 

Bad news is I am not sure what is the root cause. 

[image]
[image]

I had a hard time building when I was trying to used the cached dependencies that were in the GitHub workflow.  https://github.com/SoftFever/OrcaSlicer/blob/main/.github/workflows/build_mac_arm64.yml#L55-L59

Building everything from scratch seems to have done the trick. 

```
% clang --version
Apple clang version 14.0.0 (clang-1400.0.29.202)
Target: arm64-apple-darwin22.5.0
Thread model: posix
InstalledDir: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin
```
```
% xcodebuild -version
Xcode 14.2
Build version 14C18
```
```
% openssl version
OpenSSL 3.1.2 1 Aug 2023 (Library: OpenSSL 3.1.2 1 Aug 2023)
```

Please let me know what details you may need to track this down.

ammmze, 2023-08-05:
On mobile right now, but Interesting, I was able to build as well but I still had the issue. Going back to https://github.com/openssl/openssl/issues/20937#issue-1704871000 they mentioned it being an issue on apple clang 14.0.3 which I  have. They mentioned rolling back to clang 14.0.0 (what you have) and the issue went away. So I guess we need to see how we can build it on 14.0.0.

ammmze, 2023-08-05:
Looks like there is a setup code action we can use to set xcode version.

https://github.com/openssl/openssl/issues/20937#issue-1704871000

samwiseg0, 2023-08-05:
> On mobile right now, but Interesting, I was able to build as well but I still had the issue. Going back to [openssl/openssl#20937 (comment)](https://github.com/openssl/openssl/issues/20937#issue-1704871000) they mentioned it being an issue on apple clang 14.0.3 which I have. They mentioned rolling back to clang 14.0.0 (what you have) and the issue went away. So I guess we need to see how we can build it on 14.0.0.

I checked the github build log and it looks like it is already using `14.0.0`.
[image]

ammmze, 2023-08-05:
Hmm...I wonder if the deps were also built on 14.0.0. It looks like the deps are built separately and then extracted during the studio build.

samwiseg0, 2023-08-05:
That was my thought as well. I tried to use the prebuilt dependencies but that did not work out. I kept getting build errors.

ammmze, 2023-08-05:
I see there is a relatively new GitHub workflow to build dependencies and that looked to use 14.0.0, but I'm not sure those are used here. The date for deps used in the beta are from 2 days ago and the most recent deps build was like 2 weeks ago.

samwiseg0, 2023-08-05:
Yep. Looks like they were all built 3 days ago but not by the gihub workflow. 

https://github.com/SoftFever/FileSharing/releases/tag/2023.08

SoftFever, 2023-08-05:
No, the dependencies were not built by the CICD.
I’m currently away from my computer, so I can’t check the version at the moment.

SoftFever, 2023-08-05:
> I have good news and bad news.
> 
> Good news is after building dependencies and OrcaSlicer from source the issue is fixed.
> 
> Bad news is I am not sure what is the root cause.
> 
> [image] [image]
> 
> I had a hard time building when I was trying to use the cached dependencies that were in the GitHub workflow. I suspect these may have been built on macOS 12?https://github.com/SoftFever/OrcaSlicer/blob/main/.github/workflows/build_mac_arm64.yml#L55-L59
> 
> Building everything from scratch seems to have done the trick.
> 
> ```
> % clang --version
> Apple clang version 14.0.0 (clang-1400.0.29.202)
> Target: arm64-apple-darwin22.5.0
> Thread model: posix
> InstalledDir: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin
> ```
> 
>  ```
> % xcodebuild -version
> Xcode 14.2
> Build version 14C18
> ```
> 
>  ```
> % openssl version
> OpenSSL 3.1.2 1 Aug 2023 (Library: OpenSSL 3.1.2 1 Aug 2023)
> ```
> 
> Please let me know what details you may need to track this down.
> 
> Edit: To add. I did try beta2 and that one was still broken.

To use the pre-built depa binaries, one will need to place the Orca Slicer in the same absolute path as the GitHub action runner uses.
/Users/runner/ something like that, you can find it in the action log

samwiseg0, 2023-08-05:
@SoftFever Thank you for that info. That made the build succeed!

I think we are getting closer. That broke TLS again with the same error. Looks like the dependencies maybe being built with that clang version that has an issue?

```
Could not connect to OctoPrint: curl:SSL connect error:
error:1012606B:elliptic curve routines:EC_POINT_set_affine_coordinates:point is not on curve
[Error 35]
Note: OctoPrint version at least 1.1.0 is required.
```

ammmze, 2023-08-05:
I'm in the process of:

- [x] downloading xcode 14.2 (grabbed it via https://xcodereleases.com/?scope=release)
- [x] using `sudo xcode-select -s /path/to/downloaded/Xcode.app/` to temporarily switch to that
- [ ] re-building the app

samwiseg0, 2023-08-05:
The working and broken builds can be found [here](https://drive.google.com/drive/folders/1hVsEM7w1kkBaFgyssy9ATWz4inJ3voyx?usp=sharing).

ammmze, 2023-08-05:
Okay, after rebuilding the entire thing with 14.2, it looks we have success. I now get a 401 Unauthorized (which is expected given I don't have the credentials, just the url).

SoftFever, 2023-08-05:
Cool.
So, the conclusion is that both 14.0 and 14.2 work?

I just checked, and the deps binaries are built with Xcode 14.3.1 and Apple-clang 14.0.3.

I'll modify the workflow file for dependencies later so that we can use the binaries it built.
I hope it will work; I don't remember whether it's the Mac where I was having issues using the binaries built by CICD.

ammmze, 2023-08-05:
Sorry it's kind of confusing because the versions are so close between xcode and apple-clang

The gist is:
* Xcode 14.2 contains Apple-clang 14.0.0 and that DOES works 👍🏻 
* Xcode 14.3.1 (and presumably 14.3) contains Apple-clang 14.0.3 and that DOES NOT work 👎🏻

SoftFever, 2023-08-05:
> Sorry it's kind of confusing because the versions are so close between xcode and apple-clang
> 
> The gist is:
> 
> * Xcode 14.2 contains Apple-clang 14.0.0 and that DOES works 👍🏻
> * Xcode 14.3.1 (and presumably 14.3) contains Apple-clang 14.0.3 and that DOES NOT work 👎🏻

Great!

SoftFever, 2023-08-06:
@ammmze @samwiseg0 
The CICD can't reliably build deps yet.

But I managed to build the apes with clang 14.0.0 and updated them.
When you got time, do you mind to give this build a test? https://github.com/SoftFever/OrcaSlicer/actions/runs/5775802841

samwiseg0, 2023-08-06:
@SoftFever I can confirm this build works! Thank you for your efforts!

SoftFever, 2023-08-06:
> @SoftFever I can confirm this build works! Thank you for your efforts!

Cheers! All credit goes to you and @ammmze 🥇

ammmze, 2023-08-06:
Awesome! Great work everyone!

SoftFever, 2023-08-15:
My bad.
I missed this.
Let me add this.

SoftFever, 2023-08-15:
Added.
It's an amazing team work. Thank you @samwiseg0 and @ammmze 
[image]

closed 2023-09-07 by SoftFever