OrcaSlicer/OrcaSlicer #1677: Add Refresh button and right click functionality to the "Device" tab

Tdubprints, 2023-08-04:
I'm running Octoprint on an Ender 3, and I've found that the webpage can be buggy, especially the camera playback page. Occasionally the image for the camera will fail, or some other bug will appear on the UI screen (no fault of Orca). If I am viewing Octoprint on a browser, simply refreshing the page solves most issues. The problem is that the current browser window on the "device" tab doesn't have any refresh capability. 

Could you add a refresh button to the "device" tab? I've also noticed that the right click menu is missing (for options like refresh/reload or copy/cut/paste). This functionality would significantly improve the utility of the "device" tab. 

Thank you!

LovesDonuts, 2023-08-04:
F5 is refresh btw

Tdubprints, 2023-08-04:
Nice! This slicer just keeps getting better! Thanks!

ammmze, 2023-08-04:
FYI I had enabled the web view context menu as part of [another issue](https://github.com/SoftFever/OrcaSlicer/pull/1157#discussion_r1208378110) I was trying to debug. I don't think we have control over what is presented in the right click menu, but on Mac os it at least added a refresh menu item. It should be in the 1.6.4 beta.

BJam, 2023-08-06:
> FYI I had enabled the web view context menu as part of [another issue](https://github.com/SoftFever/OrcaSlicer/pull/1157#discussion_r1208378110) I was trying to debug. I don't think we have control [...]

Thanks, works great!  I see `Reload` as an option when I right click on device tab on `1.6.4-beta2`.

Tdubprints, 2023-08-08:
Right Click functionality is great! Thanks!

closed 2023-08-08 by Tdubprints

levydanqc, 2024-06-28:
Following on the last comment, even in V2.1.1 there is no option to refresh the Device tab...

LovesDonuts, 2024-06-28:
> Hi! This feature doesn't work for me - on Mac. Would it be possible to add the button to one-click refresh the device page? Or at least make ctrl+r/cmd+r shortcut working? It would be beneficial when klipper is being used. Thanks!

ctl +r is already there? just confirmed on 2.1.0 stable. F5 also works and right click to refresh also works. Also using klipper