OrcaSlicer/OrcaSlicer #1678: JavaScript: Unsupported result type

R4toLoko, 2023-08-04:
**Desktop (please complete the following information):**
 - OS: Mint Cinnamon 21.1 ( Ubuntu 22.04 equivalent )
 - Version: Any (??)

There was a Linux  system update and after that, OrcaSlicer keeps displaying this annoying error.
Updating OrcaSlicer didn't fix either.

[image]

[image]

Thank you

thushan, 2023-08-04:
Reproducible on PopOS after system update too in the morning, repro'd on v1.6.3 & v1.6.4b. 

Can't confirm this yet but could be because of the update of `libwpe`...

```
$ grep " install " /var/log/dpkg.log
2023-08-04 09:08:57 install linux-modules-6.4.6-76060406-generic:amd64 <none> 6.4.6-76060406.202307241739~1690928105~22.04~d567a38
2023-08-04 09:09:12 install linux-image-6.4.6-76060406-generic:amd64 <none> 6.4.6-76060406.202307241739~1690928105~22.04~d567a38
2023-08-04 09:09:12 install linux-headers-6.4.6-76060406:all <none> 6.4.6-76060406.202307241739~1690928105~22.04~d567a38
2023-08-04 09:09:28 install linux-headers-6.4.6-76060406-generic:amd64 <none> 6.4.6-76060406.202307241739~1690928105~22.04~d567a38
2023-08-04 09:09:53 install libwpe-1.0-1:amd64 <none> 1.14.0-0ubuntu0.22.04.1
2023-08-04 09:09:53 install libwpebackend-fdo-1.0-1:amd64 <none> 1.14.2-0ubuntu0.22.04.1
```

Spotigus, 2023-08-07:
Same on linux mint.

PendulumPit, 2023-08-12:
Same thing on the latest MX Linux

drewmoseley, 2023-08-13:
Ubuntu 22.04 as well.

byteofsoren, 2023-08-15:
Confirmed on ArchLinux as well and #396 as well.

avxmw, 2023-08-15:
Same for opensuse tumbleweed (tested 1.6.4b3).

R4toLoko, 2023-08-16:
This is also happening with BambuStudio which they have fixed according to `https://github.com/bambulab/BambuStudio/issues/1701`
@SoftFever I believe you mentioned that OrcaSlicer is no longer fully pulling BambuStudio code so perhaps this needs to be looked closely.

Thank you

SoftFever, 2023-08-21:
> This is also happening with BambuStudio which they have fixed according to `https://github.com/bambulab/BambuStudio/issues/1701` @SoftFever I believe you mentioned that OrcaSlicer is no longer fully pulling BambuStudio code so perhaps this needs to be looked closely.
> 
> Thank you

It's not clear what changes have they done to fix.
I pick a fix, it don't see the wxout error now.
But I still see this error, not sure if it's becuase I'm running in WSL 2

[image]

R4toLoko, 2023-08-22:
> It's not clear what changes have they done to fix.

I have checked it but couldn't find anything other than "the fix will applied to 1.7 Beta"
And to know that only now Linux users can have Dark Mode with BambuStudio is beyond me.

Will try to have a look with more time but if I had to guess, something changed on the code level and that is causing these annoying JavaScript messages every time.

WSL is a whole different beast, if you wanna to truly test Linux version, you are better off spinning up an Ubuntu or Mint Cinnamon virtual machine or you risk having WSL problems plus any Linux problem. 
WSL should be used as the very last resource haha

RAP2022, 2023-08-26:
Hi,

Below are the two commits that "fix" this error. It actually put this into a log file instead of dialog Box, along with other error I suppose.

https://github.com/bambulab/BambuStudio/commit/0699ee2da33408691e4299355e7a3a4596727c0c
https://github.com/bambulab/BambuStudio/commit/ca0847d0ac2f4c16f9d0f442bf7a8dad3767a13c

R4toLoko, 2023-08-27:
> Hi,
> 
> Below are the two commits that "fix" this error. It actually put this into a log file instead of dialog Box, along with other error I suppose.
> 
> [bambulab/BambuStudio@0699ee2](https://github.com/bambulab/BambuStudio/commit/0699ee2da33408691e4299355e7a3a4596727c0c) [bambulab/BambuStudio@ca0847d](https://github.com/bambulab/BambuStudio/commit/ca0847d0ac2f4c16f9d0f442bf7a8dad3767a13c)

Of course I am not so surprised with their "fix".
Thank you for sharing that.
@SoftFever are you able to include that on the next release?? Pleeeeeeease.
This pop-up is driving me nuts haha

SoftFever, 2023-08-27:
thank you all.
I can confirm this changes have been included in the latest main branch.
Can you help verify it: https://github.com/SoftFever/OrcaSlicer/actions/runs/5988653081

SoftFever, 2023-08-27:
> I have checked it but couldn't find anything other than "the fix will applied to 1.7 Beta" And to know that only now Linux users can have Dark Mode with BambuStudio is beyond me.
>
> Will try to have a [...]

Haha, I used Linux as my primary development OS for a while.
But I eventually moved back to Windows because of some games and software that only support Windows.
Still, WSL 2 has been a blessing, especially with its support for CUDA.

R4toLoko, 2023-08-27:
> But I eventually moved back to Windows because of some games and software that only support Windows.

Xbox joined the chat haha

> thank you all.
> I can confirm this changes have been included in the latest main branch.
> Can you help verify it: https://github.com/SoftFever/OrcaSlicer/actions/runs/5988653081

The changes are indeed in place with a slightly difference.
I tried to make the changes on my local and compile it but the `BuildLinux.sh` failed, the old problem and I just gave up

OrcaSlicer has:

[image]

And BambuStudio has:

[image]

RAP2022, 2023-08-30:
> Xbox joined the chat haha
>
>
> The changes are indeed in place with a slightly difference. I tried to make the changes on my local and compile it but the `BuildLinux.sh` failed, the old problem and I [...]

This is because of this commit I didn't mentioned earlier...
https://github.com/SoftFever/OrcaSlicer/commit/ec94c46665501886316934b4b9e18fcb08d510cb

I believe we should be able to close this issue.

ChaosBlades, 2023-09-10:
Fixed or at least hidden on OrcaSlicer V1.7.0 Beta Release
@R4toLoko Please close issue.

R4toLoko, 2023-09-10:
> Fixed or at least hidden on OrcaSlicer V1.7.0 Beta Release
> @R4toLoko Please close issue.

Sure thing, thank you.

closed 2023-09-10 by R4toLoko